MaslowCNC/Firmware #6: Strange Behavior

MakesmithTech, 2015-02-04:
If you run the following:
G17 G21 G40 G49 G54 G80 G90 G94 G64 P0.00
G0 Y-46.000
G2 X.900 Y45.900 I45.000 J0.000
M2
then
1, The arc is drawn verrry slowly.
2. After the arc completes, the machine wanders off.

MakesmithTech, 2015-02-04:
"Bar--
Increasing the gain in Circle helps with my bug 1 in http://makesmithtec.com/viewtopic.php?f=52&t=217 .
Patch attached. At least the machine labors less. Of course, the real fix is PID, which I will get to, but at least this helps. This needs to be double-checked, since I am using nonstandard servos.

If I'm flooding you, let me know. I'm trying hard to fix the other G2 patch, so far without any insight. It looks to me like it only appears when the arc is <<pi/2, but I'm not positive."

MakesmithTech, 2015-02-06:
The issue appears to be with the gcode itself, not with the way it is interpreted. The fact that it renders at all in GC may be considered a bug. Gcode was checked using: http://simplegcoder.com/js_editor_framework/

closed 2015-02-06 by MakesmithTech

shaughanlavine, 2015-02-06:
Yes, my bad.

shaughanlavine, 2015-02-06:
Yes, my bad.