MaslowCNC/Firmware #7: IJ/R errors not noted and rejected in G2 and G3

shaughanlavine, 2015-02-06:
The title says it all. This is as a result of my mistaken issue 6.

MakesmithTech, 2015-02-08:
Good suggestion! I made a feature request for it on github. I'm not 100%
sure I understand how to detect those errors.In fact, in my head I've kinda
convinced myself that the Gcode was ok.I can't think of a situation in
which it is not possible to define an arc given the current position, the
ending position and the direction.

What are your thoughts?

shaughanlavine, 2015-02-09:
Bar–
The arc must be a portion of a circle with the center defined by I and J, and so the check is just that the distance from the starting point to the center (which is the length of I+J) is equal to the distance from the center to the endpoint. I got clear on that seeing what was wrong with my example, so perhaps it wasn't a total loss. 

On a happier note, I've stnrted adding PID. I've got a PID version of SetSpeed, but I've stopped there because there is something I don't get in SetTarget: It looks like the arguments in the calls to SetSpeed are reversed. So why doesn't the machine run backward? I _think_ there must be a sign error elsewhere that cancels that one out, but I haven't found it yet. Any ideas, or am I just confused?
--
Shaughan

Putting Descartes before Horace:
Cogito ergo sum.
Dulce et decorum est, Pro patria mori.

MakesmithTech, 2015-02-10:
You are 100% right, that's exactly it. I'm glad you pointed that out because now I'm clear on it too. It shouldn't be hard to implement a quick check for this. 

On an off topic note I just merged your pull request so you are officially in the contributors page. Thanks for the awesome work!

mrfugu, 2017-07-25:
is this issue related?  http://forums.maslowcnc.com/t/issue-with-circles-randomly-being-cut/143/10

BarbourSmith, 2017-07-25:
No, this is ancient. Basically this was an issue where if you wrote gcode by hand and defined a circle which was physically impossible there was no error message thrown. I'm going to close this because it's a couple years old and nobody really seems to run into this much. I still think having an error would be nice tho.

closed 2017-07-25 by BarbourSmith