MaslowCNC/Firmware #56: Interpolating chain lengths long distances could lead to curved movements

brandonagr, 2016-10-14:
I just briefly browsed the firmware, looking at CNC_Functions.h move function, it appears that you are getting the chain lengths at start of move and at destination and then linearly interpolating the chain lengths over time. 

If you have a long move, could this result in a curved movement? Was wondering if you have tested to see if this is not an issue. (Maybe what is generating your gcode doesn't produce long movements?)

[image]

dinglewanker, 2016-10-26:
Yes.  I was wondering this same thing.

BarbourSmith, 2016-11-23:
I'm sorry guys for not seeing this sooner. I'm so used to being the only one in here that I didn't check to see if anyone else had tagged a bug. Thank you for noting that. It was absolutely a bug (or really more of an incomplete feature, there's still a good amount of those).

At some point I fixed the issue I believe by calling the kinematics function at each step along the line. The new version looks like this:

    
    while(abs(numberOfStepsTaken) < abs(finalNumberOfSteps)){
        
        float whereXShouldBeAtThisStep = xStartingLocation + (numberOfStepsTaken*xStepSize);
        float whereYShouldBeAtThisStep = yStartingLocation + (numberOfStepsTaken*yStepSize);
        
        kinematics.inverse(whereXShouldBeAtThisStep,whereYShouldBeAtThisStep,&aChainLength,&bChainLength);
        
        
        leftAxis.write(aChainLength);
        rightAxis.write(bChainLength);
        numberOfStepsTaken++;
        
        returnPoz();
        
        delay(timePerStep);
    }

Does that look like it solves the issue to you?

BarbourSmith, 2016-12-04:
I'm going to close this issue if nobody objects. Thanks again for pointing it out.

closed 2016-12-04 by BarbourSmith