MaslowCNC/Firmware #66: Misleading title

jchristin, 2016-11-28:
The title of this repo is "Makesmith Firmware". Shouldn't it be "Maslow Firmware" instead?

BarbourSmith, 2016-11-28:
Good spot! I thought I had caught all of those, but I guess no. Thank you.

closed 2016-11-28 by BarbourSmith

davidelang, 2016-11-28:
could you please make a branch of the repo for the makesmith rather than just 
abandoning it completly?

David Lang

BarbourSmith, 2016-11-28:
Yes, absolutely we can make a branch if you want to keep working on the makesmith version. I am not going to abandon the Makesmith version at all, I intend for everything to be backwards compatible. 

I've written the GearMotor library which controls the Maslow motors to use the same interface as the Servo library which controls the Makesmith motors and the Maslow control PCB can actually be used to control a Makesmith machine :-)

The big thing that prevents you from using the firmware to control a makesmith right now is the Kinematics library which relates motor movements to XYZ movement. I would like to provide a couple different kinematics layouts which the user will be able to choose from in settings (standard (like Makesmith), Hanging (like Maslow) and maybe a robotic arm kinematics also. That way the software can be used to control as many types of machine as possible. 

If you are looking for the last version of the firmware I have tested on the Makesmith, check out the releases here https://github.com/MaslowCNC/Firmware/releases