MaslowCNC/Firmware #116: Improve Execution time of Gcode

BarbourSmith, 2017-02-13:
Right now there is a 60ms delay between when each line of gcode is executed which causes a noticeable pause. For large parts it is not a huge issue, however when cutting intricate parts or if a circle has been interpolated as a large number of small lines it can slow the cutting process down.

Grbl talks about their solutions to a similar problem here: https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

BarbourSmith, 2017-02-14:
I can knock the execution time down to ~30ms by sending two lines at the same time. One is read immediately and the second one sits in the serial buffer and is immediately available when the first one finishes executing. That isn't a stable system for cutting things because the serial buffer can behave erratically from platform to platform. Basically the issue is with how fast a line can be written to the serial port.

Two part solution: 1) Read or write one character at a time from the serial port on an interrupt so that instead of waiting to read or write a whole line, the delay is split up into many small delays
2) Use a higher baud rate. Not yet, because it's much harder to get the connection to behave at higher baud rates across multiple platforms

davidelang, 2017-02-14:
you really need to add an extra buffer on the arduino so you have

serial input > buffer1
executing buffer2

then when it's finished executing buffer2 it waits until a full line is received 
and rotates the buffers

ideally set this up so there can be N buffers.

another approach is to recognize that gcode commands can be combined on a line 
and create a largish circuilar buffer, appending new characters that arrive to 
the end of the buffer and converting newlines to spaces.

David Lang

On Mon, 13 Feb 2017, BarbourSmith wrote:

> Date: Mon, 13 Feb 2017 16:19:35 -0800
> From: BarbourSmith <notifications@github.com>
> Reply-To: MaslowCNC/Firmware
>     <reply+0021b24ff5cbbd937374797661266ae4938ff176a8b9d2e292cf0000000114ba0f1
>     792a169ce0c5c2876@reply.github.com>
> To: MaslowCNC/Firmware <Firmware@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: Re: [MaslowCNC/Firmware] Improve Execution time of Gcode (#116)
> 
> I can knock the execution time down to ~30ms by sending two lines at the same time. One is read immediately and the second one sits in the serial buffer and is immediately available when the first one finishes executing. That isn't a stable system for cutting things because the serial buffer can behave erratically from platform to platform. Basically the issue is with how fast a line can be written to the serial port.
>
> Two part solution: 1) Read or write one character at a time from the serial port on an interrupt so that instead of waiting to read or write a whole line, the delay is split up into many small delays
> 2) Use a higher baud rate. Not yet, because it's much harder to get the connection to behave at higher baud rates across multiple platforms
>
>

closed 2017-05-27 by BarbourSmith