MaslowCNC/Firmware #140: Oval sleds

BarbourSmith, 2017-03-22:
It seems like everyone who is cutting their first sleds are getting an oval shape instead of a circle. This probably has to do with something I broke when setting up the math to allow for changing the dimensions of the machine in settings.

blurfl, 2017-03-24:
I found that the settings values were dramatically wrong. I used a pencil stub instead of a router bit :) but I was getting ovals until I measured the motor offsets from the workspace and sled geometry. Instant circle, accurate to within a millimeter! 
Before fixing the firmware, let's add a 'calibration' page between building the frame and cutting the first pieces...

davidelang, 2017-03-24:
how far off were they?

can you do some experimentation and see how close the settings need to be to the 
correct value to be 'close'?

I expect that you can be off by a few cm (inches) and be pretty darn close.

blurfl, 2017-03-24:
The 'before' values were the defaults created the first time GroundControl is opened:
--------------------------
motoroffsetx = 270
motoroffsety = 463
sledwidth = 310
bedwidth = 2438.4
openfile = 
sledheight = 139
zaxis = False
sledcg = 79
bedheight = 1219.2
zpitch = 20
--------------------------
After I input my measurements:
--------------------------
motoroffsetx = 285
motoroffsety = 424
sledwidth = 310
bedwidth = 2438.4
openfile = 
sledheight = 139
zaxis = False
sledcg = 65
bedheight = 1219.2
zpitch = 20
--------------------------
Note that I made a circular sled using shop tools, so my sled width and sled height match the defaults. I think this means that those measurements didn't enter into the changes I saw.

 Before the changes, the round sled file plotted 435mm wide and 466mm high. After I measured, it plotted 440mm wide and 445mm high.

 There is probably an opportunity for me to tighten up my measurements, I 'eyeballed' the measurement, and to the center of the motor shaft instead of to the top of the sprocket. I'm not sure just where the proper point would be there.

I don't have an 8.5x11 sled to test with, but the motoroffset values are in the tens of millimeters and that seems to be enough to make a visible difference.

davidelang, 2017-03-24:
can you try tweaking the CG value and see how sensitive it is to that?

blurfl, 2017-03-24:
I’ll do that, but first I tweaked the motoroffsety so that it measures from the center of the chain link pin at the top of the sprocket, now 434mm instead of 424mm. The sled circle now plots equal in the x and y directions, a more perfect circle. I think that means that we should measure from the spot that Bar indicated as the starting point for measuring the chain. FWIW, the measurement from this point to the surface that the motor mount attaches to is a constant and it would be easier for folks to measure from the edge of their workspace to that mounting surface than to a link pin in a chain...

 So that I’m sure to test what you want to see for the CG value, give me an idea of the values to try. 

 I’m currently attaching the chain to the third hole up from the sled - the sled hangs most nearly vertical when I hold it by the chains. It just occurred to me, should I instead try to get it to hang parallel to the workspace? Would this affect friction? Lots to explore!

BarbourSmith, 2017-03-24:
I played around with this too and I found I was getting even circles on my big machine, but very eliptical circles on the temporary frame. After checking all my settings carefully I realized that the chains being slanted (angled in toward the bottom corners of the plywood from the motors) was creating an optical illusion which was throwing off my measurements unless I was careful. I measured it several times and kept getting slightly different numbers (through human error, not because the dimensions were really changing).

I agree with @blurfl that we need to add a clear step to the instructions where you enter the measurements of your machine. The defaults are correct for my router on the full size machine, but they will be different for almost everyone.

In the interest of making the measurements easier during that step, I switched from measuring the width of the wood and then the motor offset from the corner to measuring the distance between the motors directly. I think that a one step process has less chance for error. 

Using the new version I was able to get clean circles on both machines.

The new version are available in PR #141 in firmware and in PR [#122](https://github.com/MaslowCNC/GroundControl/pull/122) in Ground Control. 

Because the change is in both the firmware and Ground Control we have to release them at the same time. Maybe it is time we switched to a version system instead of directing people to the latest firmware.

BarbourSmith, 2017-05-26:
We still have calibration issues, but this issue is pretty outdated and not very specific so I'm going to close it. We should re-open specific issues as we encounter them.

closed 2017-05-26 by BarbourSmith