MaslowCNC/Firmware #144: 'Calibrate Chain Length' motor speed

blurfl, 2017-03-25:
Currently runs at 500mm per minute. Can this be faster? How often is this calibration needed, would users do it more often if it took less time?

BarbourSmith, 2017-03-25:
Yes, it can absolutely be faster. I was having issues with my chains becoming tangled when it ran full speed, but I think we can probably go faster without problems.

blurfl, 2017-03-25:
If I had a couple more pieces of 'stretchy string', I'd use them to pull the end of the chain toward the opposite bottom corner, or something. I'm doing that now with garden twine, but some tension would be good.

BarbourSmith, 2017-03-25:
In theory calibrating the chain lengths is something we *should* only have to do once when we setup the machine. While debugging it comes up all the time tho, so it's worth thinking of better ways.

blurfl, 2017-03-25:
I guess the gearing is tall enough that gravity won't unwind it if the sled hangs with the power off. Do you leave your sled hanging, or rest it on something between sessions?

BarbourSmith, 2017-03-25:
I leave it hanging unless I'm working on something. There is also a worm gear in there that you can't back drive so the motors are locked when powered down so in theory you can finish cutting something, power everything down, walk away, and then power everything back up and it will all be ready to go. That seems to be true unless you pull the USB cord while the machine is running, then you need to redo calibrate chain lengths.

blurfl, 2017-03-25:
Good to know, thanks.

closed 2017-04-11 by blurfl