MaslowCNC/Firmware #148: Fix watchdog

BarbourSmith, 2017-03-26:
I think it is possible to get in a position where the machine is ready to receive a new command, but Ground Control thinks it is not and so the UI becomes unresponsive. Clicking buttons won't do anything because Ground Control is waiting for a 'gready' command from the machine, and the machine thinks it has already sent the command so it's waiting for the next instruction.

I added a watchdog that sends the ready command every second, but that seems to have broken everything and caused very erratic behavior on the machine side. I need to figure out why and add a similar mechanism.

johnbenweber, 2017-03-30:
Yes, I've found that the command sending is pretty finicky when you start changing stuff which is why I haven't implemented the gcode handling class I wrote. I think we need some more concrete communication instead of depending on kivy to do something when it detects variable changes. Depending on that functionality also makes the code flow reasonably hard to follow. I found that you can dispatch arbitrary events which you can bind to in other places. This seems to me to be more explicit and self documenting.

Back on topic, if you send multiple ready signals it makes sense that GroundControl won't be able to keep track of which command it was that finished triggering the ready message.

BarbourSmith, 2017-03-31:
I agree 100%

johnbenweber, 2017-03-31:
I think this may take a bit of work and testing, could you create a dedicated branch?

BarbourSmith, 2017-04-01:
Absolutely. It's [here](https://github.com/MaslowCNC/GroundControl/tree/investigate-watchdog)

BarbourSmith, 2017-04-11:
Added in #168 
@johnbenweber There is still more that can be done to improve the serial communication system, but I'm going to close this issue because the original problem of fixing the watchdog function is done.

closed 2017-04-11 by BarbourSmith

johnbenweber, 2017-04-13:
@BarbourSmith Looks like I've missed some fun while I was away. I'm working on getting some bigger motors to get my setup running. Everything is in place really, I just need more torque.

BarbourSmith, 2017-04-13:
There's plenty of fun still to be had 😀 

Did you get the motors sorted? We just got our big shipment so we've got plenty of them now. Email me your address and I'm happy to send you some! ✈️