MaslowCNC/Firmware #158: Add software support for limit switches

davidelang, 2017-03-31:


BarbourSmith, 2017-03-31:
Great suggestion! I hadn't thought about the idea of limit switches on the z-axis

davidelang, 2017-03-31:
limit switches an all motors would be great, but they are harder to do than on 
the Z axis.

you could detect the real '0' point on the z axis if you have a sensor with one 
lead on the motor/bit and the second lead a flat metal plate that you slip under 
the router

for the other two motors, you could create a swinging device that rotated with 
the chain and had a switch for when the sled came up to it. This would allow it 
to 'retract to zero' on each side.

This is more complicated than the current mechanism of just measuring and 
assuming that things don't move while it's shut off, and it requires additional 
finicky wiring, so I'm not at all pushing for this to be part of the basic 
machine design, but it could be a cheap and useful add-on if the software can 
support the concept.

You are working on higher priority things right now, I mostly created this to 
act as a placeholder for later.

David Lang

johnbenweber, 2017-04-03:
I think you could do the x-y limit without switches. Because we have feedback you can just detect motor stall. If you had some wooden pieces attached to the bed you could run the sled into it until both motors stall and call that the x-y home. See rough diagram:
[image]

BarbourSmith, 2017-04-03:
At the very least having stall (or even just sudden changes in the torque load) detection could prevent situations where the machine tries to break itself. The motors are strong enough that they can break the frame of the machine, but if we monitor the torque each motor is putting out we can see when something is going wrong.

Good suggestion.

johnbenweber, 2017-04-03:
Ah, well that would be bad. I didn't realize the motors were that strong.

jbarchuk, 2017-04-04:
johnbenweber commented a day ago If you had some wooden pieces attached to the bed you could run the sled into it until both motors stall and call that the x-y home. See rough diagram:

I've been wondering how much resolution the torque detector has, what kind (the electronics, just curious), how fast it updates, part to part repeatability... Calibration as described would be awesome but I have no idea if it's vaguely doable or even useless.

davidelang, 2017-04-04:
The torque detector is not a separate component, it is a calculation.

What is happening is that the processor is configured to move the motor at a 
given rate (based on feedback from the sensor), and to do so it sets the motor 
to run at in a given direction at a given duty cycle, the higher the duty cycle, 
the faster it will move. If it is not moving as fast as it is supposed to, it 
increases the duty cycle. If it is moving faster than it is supposed to, it 
decreases the duty cycle.

The current duty cycle setting is a very good approximation for the amount of 
torque the motors are applying

So if you have not instructed the motors to change what they're doing, and the 
duty cycle increases suddently, then you have hit something and are trying to 
power through it. This will happen to some degree as you are moving around and 
the bit moves from an area that was already cut to an area that needs cutting, 
but this is a small jump compared to hitting the end of the chain and trying to 
spin the sled around the motor. :-)

mrfugu, 2017-07-16:
early stages of thinking it through, but I believe that something like these [Switches]( https://www.amazon.com/Longruner-Mechanical-Endstop-Makerbot-LKB01/dp/B06XTB7WMK?SubscriptionId=AKIAILSHYYTFIVPWUY6Q&tag=duckduckgo-d-20&linkCode=xm2&camp=2025&creative=165953&creativeASIN=B06XTB7WMK) located in a box next to the Electronics with mechanical arms reaching out to some lightly sprung arm running some length of the 4 sides of the frame edge/workspace. Perhaps with cutouts for clamps for the workpiece?  Still waiting on my MaslowCNC, but these switches seem to fit the bill and it should be rather trivial to use the machine to construct arms reaching out to the work surface as opposed to running wires here and there.

davidelang, 2017-07-16:
you want to allow the sled to go off the edge of the workpiece, just not too far 
off the workpiece.

you will also have different size workpieces at different times.

you will also have problems with your paddles being stiff enough to not flex and 
light enough to move easily.

I also expect that the arms are going to get in the way of loading and unloading 
your workpiece.

johnbenweber, 2017-07-17:
@davidelang beat me to it but I'll add that I don't think finding individual edges of the work piece actually tells you much without having more information (individual chain lengths) because the motor axes aren't separable or aligned with the work piece axes. If you do know the chain lengths then you know where you are and don't need the switches anyway.

mrfugu, 2017-07-17:
david, agreed. 

I'm proposing that it might be possible via purely mechanical means, reaching around the frame and closing short throw switches located close to the electronic bundle, vs running wires out from the electronics bundle and placing switches around the frame.

krkeegan, 2017-07-18:
What about pressure sensors in the sled.  Figure out what the most level of overhang you want the sled to have and then place some sort of pressure sensor there.  Then if the sled comes off the work piece by more than you want, it stops.  This would also stop things if the sled jumped off the work piece, which you may be a nice benefit anyways.

krkeegan, 2017-12-27:
Soft and Hard limits can be checked from the realtime function.

So now the issue is just how do we allow users to define the pins and the characteristics?  How many pins could be used?  2 per axis?  and how do we define whether they are high or low?  Finally, how should users be defining this?  In precompile defines or settings?

blurfl, 2017-12-27:
I vote for precompile defines. Let’s be conservative about adding runtime settings.

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Scott Smith wrote:

> I vote for precompile defines. Let’s be conservative about adding runtime settings.

I agree with this. It's common in the 3D printing world for you to need to 
define limits like this.

Generally they allow 1 bit per axis, wire both limit switches to the same pin, 
and then figure out which one they hit by which direction they were traveling in 
when they hit the limit.

Marlin is doing some interesting work right now (derived some work that Prusa 
did with their newest printer) to support drivers that can report failed steps. 
They are adding the ability to do sensorless homing by detecting the attempt to 
move without actually moving (we were discussing this sort of thing not too long 
ago here)

krkeegan, 2017-12-27:
Cool.  1 pin per axis and using defines is easy enough.

How do people normally handle HIGH or LOW.  Should I just pick one as the fault state (HIGH)?  I could probably do something with a change in state, but that could get messy if the state ends up inverted.

krkeegan, 2017-12-27:
> They are adding the ability to do sensorless homing by detecting the attempt to
> move without actually moving (we were discussing this sort of thing not too long
> ago here)

I am still interested in trying this.  I still have concerns that given our high torque it is going to be hard to see this.  But if nothing else we can detect a complete failure as in https://github.com/MaslowCNC/Firmware/issues/277.

blurfl, 2017-12-27:
Would using the work area dimensions to calculate soft limits make sense?

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Kevin Robert Keegan wrote:

> Cool.  1 pin per axis and using defines is easy enough.
>
> How do people normally handle HIGH or LOW.  Should I just pick one as the fault state (HIGH)?  I could probably do something with a change in state, but that could get messy if the state ends up inverted.

configured in the source (along with which pin is used for what, what direction 
things actually move, etc)

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Scott Smith wrote:

> Would using the work area dimensions to calculate soft limits make sense?

Yes, but there should be an ability to have multiple sets of soft limits defined 
(or make it really easy to adjust them)

blurfl, 2017-12-27:
So, a user-editable setting that defaults to +-h/2, +-w/2?