MaslowCNC/Firmware #183: Settings should really be stored in the firmware

BarbourSmith, 2017-04-21:
The machine settings should really be stored in the machine's firmware. The firmware is connected to the machine. That the settings are not stored in the firmware causes a couple issues.

If you unplug from the machine, leave Ground Control open, plug back in the settings are not sent to the machine again. This means that the machine switches to using the default settings which is confusing and wrong.

I don't want to fix this issue until after next week's release because I just don't want to mess with the way the settings work and this is going to mess with all kinds of things. We're just starting to get things worked out, and I want at least one good week before we start changing things again. That way we can point back to version .6X as the good one.

BarbourSmith, 2017-04-21:
Partial fix added in GC PR [#219](https://github.com/MaslowCNC/GroundControl/pull/219), settings are now sent to machine when GC reconnects. 

Storing them in the EEPROM is still the right way to do things.

krkeegan, 2017-11-29:
I have been thinking about this a bit and may start to work on this if anyone has any concepts or methodologies they would like to include.

BarbourSmith, 2017-11-29:
That would be awesome!

Maybe a good "reset to defaults" function is the only feature I would want. I know we had issues in the past with things stored in the EEPROM giving people issues that looked like hardware problems, but were actually firmware

Happy hunting!

krkeegan, 2017-12-07:
I spent about a week trying to become acquainted with the grbl source code.  I have a general idea about how it works and it is pretty impressive.

One aspect I would like to copy here and am looking for feed back is the use of $ as a machine specific command.  I don't know how we started using B codes for this, it works because we don't have a B axis, but issues could arise since this design conflicts with the gcode spec.

Here is the the grbl wiki documenting their $ commands: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

At least for a start, I would like to implement the $$ and $n=val concept.  The thought is that GC can send $$ on startup, review the values returned by the firmware and then send the necessary $n=val commands to update any settings that are different.

Thoughts/Complaints?

davidelang, 2017-12-07:
This sounds like a good step forward.

BarbourSmith, 2017-12-07:
I think that sounds like a really solid plan. GRBL seems like a really really solid platform to base our work off and the more we can be compatible with it the better

krkeegan, 2017-12-20:
This is moving along well.  I think I will have something for us to test out this week.

closed 2017-12-27 by BarbourSmith