MaslowCNC/Firmware #188: 3 Axis Coordinated movement for linear interpolation

mindeye, 2017-04-28:
Currently, just X and Y axis movements are coordinated. In some cases, having all 3 axis (X, Y, and Z) moving in a coordinated fashion would lend itself to doing more complicated 3D cuts. I'm envisioning this working in the manner that the cutter moves in a straight line in 3D space from it's start point to the end point specified by the X,Y,Z coordinates.

The gcode I saw in my output from Fusion 360 looked something like this (in terms out output, not values)

G01 X1.3 Y1.3 Z1.3

Let me know if any other info would be helpful.

BarbourSmith, 2017-04-28:
Looks perfect! Thanks for filing the issue 💯 👍

krkeegan, 2017-10-13:
This is fixed by #315

closed 2017-11-28 by BarbourSmith