MaslowCNC/Firmware #189: make g0 move at the machines max speed

davidelang, 2017-04-29:
G0 is supposed to always move at the machines max speed, while G1 moves at the defined feed rate. Currently there is no difference between the two.

BarbourSmith, 2017-04-29:
Good point. The way it works right now is that G1 moves the machine at a defined speed, while G0 always moves the machine at it's maximum speed which is 48ipm. 

The real spec for G0 is to run both motors at full speed until it gets where it should be. I found that this caused undesired behavior near the edges of the 4*8 area when one motor had much more distance to cover than the other causing the sled to leave the 4*8 space. That is why I removed the original G0 code and instead do a coordinated move at 48ipm.

davidelang, 2017-04-29:
g0 is defined as being a coordinated move, and since that doesn't get you to the 
final position any slower than an uncoordinated move (you are still limited by 
the axis that takes the longest to get there), it doesn't harm anything.

I missed the code that detected the difference between g0 and g1

closed 2017-04-29 by davidelang