MaslowCNC/Firmware #190: support G91 (relative mode)

davidelang, 2017-04-29:
support G91 (relative mode) addressing.

currently if you were to do G91 G1X1Y1 G1X-1Y1 repeatedly, the result would be one line with the router moving bac and forth when it should result in a zig-zag across the work.

relative mode addressing is great for creating a chunk of gcode that gets re-used repeatedly across a workpiece.

BarbourSmith, 2017-04-29:
Very strange, I will look into this.

I was under the impression that G91 was working, we actually use it for the arrow buttons so it's at least partly working.

Thank you for the clear test case with the description of what should happen.

davidelang, 2017-04-29:
from reports after I filed this, it looks like this may be working in the 
firmware, but buggy in GroundControl

BarbourSmith, 2017-04-29:
Interesting and good to know. Do you have a way to make it not work consistently? (so I can track it down more easily)

davidelang, 2017-04-29:
at [testpattern2.nc](http://lang.hm/maslow/testpattern2.nc) has code that GroundControl claims will run off the edge of the workpieces, but actually doesn't

This also shows problems due to the lack of handling the T and M6 (tool change codes) The Spindle on/off commands (M3/M5) are also ignored

`
(Generated by Hand)

G21 G90 G40

G0 Z5
G17
G0 X-900 Y450
G91

T0 M6
M3
G0 X450
G1 Z-3
G1 X18
G0 Z5
G0 X-18 Y-900
G1 Z-3
G1 X18
G0 Z5
G0 X-900 Y900
G1 Z-3
G1 Y18
G0 Z5
G0 X900 Y-18
G1 Z-3
G1 Y18
G0 Z5
G0 X-432
M5
T1 M6
M3
G0 X450
G1 Z-3
G1 X18
G0 Z5
G0 X-18 Y-900
G1 Z-3
G1 X18
G0 Z5
G0 X-900 Y900
G1 Z-3
G1 Y18
G0 Z5
G0 X900 Y-18
G1 Z-3
G1 Y18
G0 Z5
G0 X-532
M5

`

BarbourSmith, 2017-04-29:
Perfect! That's exactly what I needed. You are totally right that GC doesn't know how to render relative commands. I'll fix this ASAP.

Thanks for the find!

BarbourSmith, 2017-05-01:
Rendering support for G91 added in Ground Control pull request [#238](https://github.com/MaslowCNC/GroundControl/pull/238) so the newest Ground Control should render files using G91 correctly.

Here are the examples from above:

Zig-zag using G91 vs G90:

[image]
[image]

And the test shape:

[image]

Let me know if you still have any issues.

Good request!

closed 2017-05-01 by BarbourSmith

BarbourSmith, 2017-05-03:
@davidelang Is is ok if I used your test file as the basis of the test code that GC will run to calibrate the machine?

davidelang, 2017-05-03:
sure, let me know if you need me to do any others.

David Lang