MaslowCNC/Firmware #204: add direction option for Z axis motors

davidelang, 2017-05-02:
between different ways of attaching a screw to the router, different directions to mount the motor from, and use of belts, there are many possible reasons why the z-axis motor may need to run in the opposite direction.

If this was during the design stage, I'd suggest that the plugs be made so that reversing a plug would reverse the direction the motor spins, but we're far past that.

In any case, software is easier to change than hardware, so there should be a config option on the main board to run the motor in the opposite direction.

The same thing should go for any use of the spare axis (and ideally for the main two, but it's hard to imagine why they would need to change right now, but in the interest of future-proofing things, it would probably be good to go ahead and include them)

walter, 2017-05-03:
Background discussion can be found [here](http://www.maslowcnc.com/forums/#!/general:my-z-struggles/after-running-the-test-shap).

BarbourSmith, 2017-05-03:
Great news everyone!

This feature actually already exists, just put a negative number in for the thread pitch and it will reverse the direction of motion. 
[image]

closed 2017-05-03 by BarbourSmith