MaslowCNC/Firmware #205: add option to use any drive for any purpose

davidelang, 2017-05-02:
since things can happen to driver boards and plugs, it would be a good idea to have the ability to define any motor to be on any port so that they can be re-plugged to work around issues (or even to test things)

krkeegan, 2017-12-27:
I think we are really talking about arduino pin mapping here not just Maslow motorshield mapping.  This would enable someone to roll their own shield.

This could be done using precompile defines, which I think is appropriate and wouldn't use any additional memory.  I think someone who is remapping motors should be comfortable compiling the firmware themselves.

Otherwise, if we want we can make settings for all of the pins (encoders, direction out, and pwm) for each motor.  This would be 15 bytes for our current setup (5 more bytes for each additional motor)

blurfl, 2017-12-27:
This arose to address the situation where the right motor H-bridge burned out; one could re-map the right motor to the z-axis H-bridge. The driver chip probably wouldn't handle both left and right motors without damage. 
  Using precompile defines makes sense.