MaslowCNC/Firmware #245: Jump the version number

davidelang, 2017-06-07:
it makes for less confusion if the firmware and ground control have the same version numbers, and with development ongoing on both of them, there are unlikely to be weeks where one is changed and the other isn't (at least for a fair while)

So I would suggest that you bump the version number of the firmware to match the version of ground control in next week's update

blurfl, 2017-06-07:
If the ‘version number’ were a string, we could add sub-release codes as well. Currently it is a float that rounds to two digits after the decimal point.

davidelang, 2017-06-07:
actually, version number is a #define, which is a string substatution. It may be 
getting treated as a float when being output, but it really should be treated as 
a string.

BarbourSmith, 2017-06-08:
I think this is a great idea, including the sub version number. If you have time to throw together a pull request to make sure the version number is treated as a string I will happily merge it. At the very least I'll make the version numbers sync up with next week's release

blurfl, 2017-06-08:
Ok, I’ll do it later this morning. But first:
 What numbers to use? I’ve no experience here. Does one use the PR# as the subversion number, or maybe the sub-subversion number? Just a sequence, zero-based?
 What about moving the version number to near the top of the .ino file so it’s easily seen?
 Can GC have it’s version number in a place other than data.py; where would that be?

BarbourSmith, 2017-06-08:
I was imaging it as a sequence just so that we could check that we're using the same build when we're testing mid week. Each week we would start the week at v0.75.0 and then increment it to v0.75.1 with each notable change. That seems straight forward to me, but I also don't have a lot of experience here.

I fully support moving the version number in the .ini file

The version number is in data.py because the information in data.py is accessible to most of the objects in the program, but maybe we don't need that. Maybe it could be in main.py?

davidelang, 2017-06-08:
what we want to do is to tag each week in git (as you are doing) and then "git 
describe" does a good job of identifying exactly how far you are from the 
release

https://stackoverflow.com/questions/3300746/deriving-application-build-version-from-git-describe-how-to-get-a-relatively
http://www.soliantconsulting.com/blog/2016/07/versioning-with-git-describe

google for 'use git describe in a build' or similar and you will find lots of 
examples.

blurfl, 2017-06-08:
Sounds good, is there a way to automate that for the Arduino GUI?

davidelang, 2017-06-08:
that I don't know. but even if it's done manually (shell script for linux, batch 
file for windows), it would be better than a value that needs to be edited 
manually.

blurfl, 2017-06-08:
Would you be willing to write the scripts?
Who would run the script, at what point in the fork/branch/PR sequence?

davidelang, 2017-06-08:
so the script for linux (and changes to use it) would be something like:

diff --git a/cnc_ctrl_v1/CNC_Functions.h b/cnc_ctrl_v1/CNC_Functions.h
index a2a9a10..2d5989b 100644
--- a/cnc_ctrl_v1/CNC_Functions.h
+++ b/cnc_ctrl_v1/CNC_Functions.h
@@ -20,7 +20,7 @@ libraries*/
  #include "Kinematics.h"
  #include "RingBuffer.h"

-#define VERSIONNUMBER 0.73
+#include version.h

  bool zAxisAttached = false;

diff --git a/cnc_ctrl_v1/makeversion.sh b/cnc_ctrl_v1/makeversion.sh
new file mode 100755
index 0000000..9f8f17c
--- /dev/null
+++ b/cnc_ctrl_v1/makeversion.sh
@@ -0,0 +1,3 @@
+#!/bin/sh
+echo -n "#define VERSIONNUMBER " >version.h
+git describe --tags >>version.h

it should be very easy to do a similar batch file, but I don't have a windows 
machine available to test on.

I just did a git pull, and since there are no changes since yesterday's release, 
the result is:

#define VERSIONNUMBER v0.73

just before I did the pull, the result would have been

#define VERSIONNUMBER v0.72-22-g132c76e

saying that this was based on v0.72 with 22 new commits, and commit 132c76e was 
the version being compiled.

this script should be run just before compiling. It isn't involved at all when 
doing a PR

blurfl, 2017-06-08:
diff —git doesn’t seem to be a valid option for diff.

Would you be willing to write the scripts to do this? It’s more than I had in mind when I volunteered to create the PR 😐

davidelang, 2017-06-08:
the script is:

#!/bin/sh
echo -n "#define VERSIONNUMBER " >version.h
git describe --tags >>version.h

to use the result, you need to replace the line:

#define VERSIONNUMBER 0.73

with the line

#include version.h

the file version.h is created by the script.

on windows, try making a .bat file that contains.

echo | set /p dummyName="#define VERSIONNUMBER " >version.h
git describe --tags >>version.h
k

blurfl, 2017-06-08:
When run as a script from within cnc_ctrl_v1/ on OSX, version.h contains two lines:

-n #define VERSIONNUMBER 
v0.73-1-g0bca9ab

Changing the first line to    #!/bin/bash    , version.h contains:

#define VERSIONNUMBER v0.73-1-g0bca9ab

This is because /bin/sh on os x runs bash with a limited command set.

Beyond that, the lack of quotes causes an error in the compiler. I haven’t tested on Windows yet.

> On Jun 8, 2017, at 2:22 PM, David Lang <notifications@github.com> wrote:
> 
> #include version.h

davidelang, 2017-06-08:
oops, I didn't realize that the -n was a bashism

changing it to

#!/bin/bash
git describe --tags | while read tag
do
   echo "#define VERSIONNUMBER \"$tag\"" >version.h
done

should work and include the needed quotes.

blurfl, 2017-06-08:
That does the trick.
 It’ll be a while before I’m able to try the Windows .bat file, maybe someone else will have  chance? Is the single line ‘k’ part of the .bat script? Not familiar with that…

davidelang, 2017-06-09:
no, the k was a typo (when I reply via e-mail, control-k deletes a line, and if 
I release the control before the k and hit send too quickly, it leaves a stray k 
in the e-mail)

blurfl, 2017-06-09:
I've realized that my Windows setup does not have a command line git installed. Someone else will need to test that part. Perhaps the linux version of the git tools will run using the recent Windows linux setup.

mrfugu, 2017-07-25:
Housekeeping: this issue apparently can be closed.

closed 2017-07-25 by BarbourSmith