MaslowCNC/Firmware #257: Motor with different encoder moving to fast

gilbertf, 2017-07-01:
Hello,
at the moment I'm trying to use the maslow firmware with an different motor. The specifications are as follows:
512 CPR, ~44,73:1 gear ratio

The first problem is in MotorGearboxEncoder.cpp:
float RPM = (7364.0*distMoved)/float(timeElapsed);  //6*10^7 us per minute, 8148 steps per revolution

What about calculating the 7364.0 like follows:
#define RPMSTEPS       60000000.0 / ENCODERSTEPS //60*10^6 us per minute

Does this introduce to much computations such that we need to do the calulation at compile time?

Furthermore why is 7364.0 used instead of the actual 7363.8?

Having changed this my motor still turns too long. E.g. The button for one degree rotation in the chain lengh calculation menü gives me one full turn and PID does not seam to work since changing the input voltage changes the number of turns.
Do you have some suggestions where to look next? The motor test function is giving reasonable puls count values.

gilbertf, 2017-07-02:
Making some progress now and the motor is somehow moving correctly. Have found a blog entry mentioning the tune-velocity-PID-branch branch, that looks like to be unavailable now. Anybody can help to access the files.
Thanks a lot

johnbenweber, 2017-07-09:
There are options in GroundControl to specify the number of encoder steps/revolution as well as the number of gear teeth. You need to open the "Settings" menu and then select "Advanced Settings" at the top.

BarbourSmith, 2017-07-11:
Hey @gilbertf sorry for the slow response, we've been frantically filling all the orders this week. @johnbenweber knows the most about using different types of motors. The tune-x-PID branches are pretty far in the past at this point, but I think playing around with tuning the PID controllers is a good idea.

You can find the values you would want to change (_Ki, _Kp, _Kd) in the files Axis.h (the position PID controller) and MotorGearboxEncoder.h (The velocity PID controller) 

Let me know if you have any specific questions, but I'm no PID tuning expert myself, the values we are currently using are probably pretty far from optimal, so if you find a good system to derive better ones, let me know!

BarbourSmith, 2017-11-28:
Is this still an issue?

krkeegan, 2017-12-27:
I don' think this is an issue anymore and we aren't receiving any additional comments.

closed 2017-12-27 by krkeegan