MaslowCNC/Firmware #262: Right Motor Strange Behavior

BarbourSmith, 2017-07-16:
We are seeing a lot of strange behavior with firmware .79 and .80 with the right motor not behaving correctly. The issue seems to be widespread but doesn't happen for everyone. I haven't yet been able to replicate it which is making it tough to track down.

If everyone would post the behavior you are seeing, we can compare and see what features are common to everyone's experience of the issue.

BarbourSmith, 2017-07-16:
The first thing I would ask people to try is this firmware here: https://github.com/MaslowCNC/Firmware/tree/explore-optimization-issues This on e tests to see if maybe the compiler is deciding that some functions are not needed (even if they are) and removing them which could explain why some are seeing the issue while others are not

BarbourSmith, 2017-07-16:
The next thing to try is this firmware here: https://github.com/MaslowCNC/Firmware/tree/right-motor-option-two-v1.1 As I've been poking around I'm starting to think that we might actually be running out of memory space for global variables as the program runs. This branch tests to see if freeing up some space by hard coding the pinlayout and reducing the buffer size fixes the issue. Because the pins are hard coded, this version will only work with PCB v1.1 and not with the beta boards

BarbourSmith, 2017-07-16:
I still was unable to make the same issue that everyone else has been seeing but I was able to get my right axis to behave strangely and then address the issue by freeing up some more memory. I think we're seeing the issue on the right axis, because the left axis is declared first, so it ends up with a preferential space in memory. Swapping the order of lines 128 and 129 in CNC_function.h

 `Axis leftAxis (ENC, IN6, IN5, ENCODER3B, ENCODER3A, "Left-axis",   LEFT_EEPROM_ADR, DISTPERROT, ENCODERSTEPS);

Axis rightAxis(ENA, IN1, IN2, ENCODER1A, ENCODER1B, "Right-axis", RIGHT_EEPROM_ADR, DISTPERROT, ENCODERSTEPS);
`

I was able to move the problems I was seeing from the right motor to the left motor which makes me think it's not an issue particular to the pins the motor is connected to. 

A memory allocation issue would explain the fact that we're seeing the issue manifest differently for different people because the Arduino compiler will be different on different platforms (I think).

BarbourSmith, 2017-07-16:
Finally here is a branch which frees up some memory, but will work on both the beta arduino shields, and the v1.1 shield: https://github.com/MaslowCNC/Firmware/tree/reduce-buffer-size .

If this branch works, we can merge it into master and fix the problem for everyone while I look into more ways to reduce our memory use.

mribble, 2017-07-17:
Thanks for looking into this so quickly.  The issue I had was when the second motor was supposed to run it would just hum, but not turn.

Here are the things I tried:

1. I installed the original 0.80 software where I was having issues with my workstation which had newer version of the arduino software.  This fixed my problem.
2. I ran your explore-optimization-issues branch on my laptop and it didn't fix the issue.
3. I ran your latest reduce-buffer-size  branch on my laptop and it fixed the issue.
4. I updated my laptop to the latest Arduino software and that fixed the issue even with the old 0.80 software.  I updated the IDE from 1.8.0 to 1.8.3.  The IDE doesn't really matter, but the AVR Boards toolchain does and this update also upgraded this to 1.6.19 (sorry, I didn't check what version I was using before the upgrade).

From this it seems like your root cause about the problem being sram related seems correct.  I looked at your branch were you changed the buffer size from 256 to 128 and I can't really say if that's safe without reviewing the rest of your code, but you are the expert and seem to think that's safe so I say go for it.

It also might be the case that the people seeing the issue are using older versions of the arduino software (at least that was true for me).  So if others are having this issue and for some reason don't want to try this branch you could suggest updating to the latest arduino avr board toolchain library.

BarbourSmith, 2017-07-17:
@mribble Thanks for the through break down of what worked and what didn't, I feel much better about my sram hunch now. Shrinking that buffer is absolutely safe right now (we don't even use it). If that solves the issue for everyone I'll give everything a clean up because I can think of quite a few places we could save some ram easily.

blurfl, 2017-07-17:
Recent development has introduced a lot of string handling. On my system, global variables use 3399 bytes (44%) of dynamic memory. Using the F() macro to move static strings into PROGMEM, frees up 1012 bytes of dynamic memory. This would provide more stack space.

blurfl, 2017-07-17:
@BarbourSmith would you like a PR that moves the static strings?

BarbourSmith, 2017-07-17:
@blurfl I would love that. I think we can even shrink them also. Most of the static strings are _axisName instances, right? Instead of naming them "Left-Axis"...etc we could name them "1" or "L" and not really loose any information, right?

blurfl, 2017-07-17:
I’ll do it. Which branch would you like it based on? 
 If the strings are stored in program memory, their length doesn’t matter as much. Long strings do use processor time and clog up the terminal and log, though.

sblite, 2017-07-17:
Just tested the latest firmware and whilst the right motor still makes a buzzing noise it does turn and seems to have calibrated the chain correctly

davidelang, 2017-07-17:
but the names of the strings don't matter, the names are only in the source 
code, they are not part of the resulting binary image that gets flashed.

mribble, 2017-07-17:
Davidelang is right.  Let me explain this a little more since I could see where this is confusing for someone with interpreted language experience like javascript or non-compiled python.

For languages like javascript the variable names are used the first time the code is run.

The arduino ino files are compiled down to machine codes since it's basically C++.  In short the variable names completely go away and you are just left with offsets into different parts of memory.

The F() macro is useful for AVR arduino chips.  AVRs like the 2560 don't have an instruction to directly access data from the flash chip.  This means the compiler can't do it.  So instead Arduino created this macro that does this conversion.  It's slower than storing strings in sram, but most of the time saving sram for extra processing cycles with strings is a good tradeoff.  The F() macro doesn't do anything if you are using an ARM arduino variant because the ARM instruction set has an instruction to directly fetch data from flash.

blurfl, 2017-07-17:
I’m referring to the content of the static strings, all of which happen to be unnamed. See PR#263 changed files for examples. Using the F() macro on the many strings in print statements frees up just over 700 bytes when measured in the firmware running on the Mega. Many of the problems we’ve seen are consistent with stack crashes, this might address that.

BarbourSmith, 2017-07-17:
Thanks @blurfl . Great work, great solution!

davidelang, 2017-08-03:
I think we need to do some testing to see if we have a hardware/crosstalk problem happening.

As I understand the problem, we see interactions between running the right motor and the Z axis.

This could be caused by the motor pulses being sent to one motor showing up on the encoder lines for the other motor. This would be dependent on the exact layout of the wiring (how close are the different wire bundles run) which would explain why some people see this as a big problem, and others don't.

would it be hard to setup a test where one motor is exercised at full power (or potentially at different power levels to check that it's not something sensitive to the PWM power level) while monitoring the encoder lines for the other motors?

If moving one motor generates pulses on the encoder lines for the other motors, this could be the cause of some of this behavior.

The good news is that a lot of this problem can be dealt with by adjusting the cable layout.

blurfl, 2017-08-03:
The PWM signals for the three motors are assigned to gpio pins in lines 130-132 of CNC_Functions.h the right motor is ENA, the z motor is ENB and the left motor ENC. 
Here are the traces for the [right motor](https://user-images.githubusercontent.com/1851315/28948617-b6351e6a-786b-11e7-810b-aa087ef13811.jpg), the [z motor](https://user-images.githubusercontent.com/1851315/28948619-ba0699b0-786b-11e7-9b2a-cb33c1e022e0.jpg) and the [left motor](https://user-images.githubusercontent.com/1851315/28948621-bdcfb270-786b-11e7-9980-71770021a6a9.jpg).
You can see from the pictures that if there is crosstalk, it is more likely to be the right and z motors.
 The Test_Electronics_Frimware sketch in the Firmware directory would make a good starting point for a crosstalk test, though.

davidelang, 2017-08-04:
I think there's also a strong potential for crosstalk between the cables. If you put the control panel to the left of the machine, the right and Z cables will naturally get placed next to each other.

blurfl, 2017-08-04:
On my rig, the Z cable pretty much dives down vertically toward the sled and the right motor cable takes off horizontally to reach the motor. But as you say, it should be simple to make sure that the cables are not routed parallel. Are you able to run your cables close together for a test?

BarbourSmith, 2017-08-04:
My hunch is that this has to do with the PID tuning again 😕

blurfl, 2017-08-04:
Since you brought up the “P” word... I’ve been playing around with #277 and I notice that there’s often a _pidOutput even when a motor is stopped. Do the motors share any PID values, or are they totally isolated? If isolated, should the z axis have different default values than the motors?

davidelang, 2017-08-04:
On Fri, 4 Aug 2017, BarbourSmith wrote:

> My hunch is that this has to do with the PID tuning again 😕

PID tuning would not make one motor move when you send a command to a different 
one, that almost has to be crosstalk. (moving one motor generates what looks 
like a signal on the encoder lines of a different motor, and then the PID loop 
jumps in to 'correct' the position of the second motor, but since it didn't 
actually move it moves too far and has to correct itself)

davidelang, 2017-08-04:
On Fri, 4 Aug 2017, Scott Smith wrote:

> should the z axis have different default values than the motors?

Yes, the motors, encoders, and loads are enough different that the Z should have 
different PID tuning than the other two motors.

closed 2017-11-28 by BarbourSmith