MaslowCNC/Firmware #265: Firmware 0.81 Twitchy movement

jbnimble, 2017-07-21:
After changing from Firmware 0.80 to 0.81 the sled keeps twitching, especially when moving the Z-Axis. It looks like a small circular motion or back and forth twitch. The motion is not much, but it definitely has a wiggle where I did not notice it before.

May not be relevant or a change from previous, but also noticing the Z-Axis moves in short bursts when it is moving up or down.

BarbourSmith, 2017-07-21:
Hmmmm...that sounds like exactly the memory overflow issue 0.81 was supposed to fix. Everyone else seeing this?

krkeegan, 2017-07-21:
My XY movement is minimal now when the Z-axis is moving, I had major issues with 0.80.

But I do agree that the Z movement is very sporadic.  When retracting in particular it stutters a lot which seems really odd particularly since I have been trying to retract at the fastest speed possible.

BarbourSmith, 2017-07-22:
I think this has to do with the PID values being tuned for the XY motors which is resulting in the z-axis expecting more load  that its seeing so if overshoots, then waits, then over shoots, then waits...etc

JDogHerman, 2017-07-31:
This issue should be fixed with version 0.82.

This issue can be closed.

krkeegan, 2017-07-31:
The PID fix also solved this for me.

closed 2017-08-01 by BarbourSmith