MaslowCNC/Firmware #266: Coordinated Move Ignoring Z-Axis Position

krkeegan, 2017-07-21:
When retracting the z-axis is followed by an XY coordinated move, it seems that frequently, but not always, the coordinated move doesn't wait for the z-axis to complete its operation before continuing.  This results in a squiggly cut between the two points.

Here is some G-Code where I see this happening:
The relevant g-code is:
G1 Z-0.75 F35
X-0.6872 Y-0.0042
Z0.2 F30
(the coordinated move below often happens before the above command completes)
M9
G0 X-0.1116 Y0
G1 Z0.2 F35
Z-0.1875
Y0.1039 F15
G3 X-0.1138 Y0.1314 I-0.175

BarbourSmith, 2017-07-22:
Perfect! (I mean perfect description of the issue, the problem is a problem 😀 )

I'll see what I can figure out.

krkeegan, 2017-07-24:
The following forum post has more information on this issue.  Sorry some things like images can't be posted here.

[http://forums.maslowcnc.com/t/z-axis-not-fully-retracting-before-moving/175/6]

krkeegan, 2017-07-25:
I think I found the bug that caused the initial problem.  Not sure if it would have caused the huge cut though.  Let me test it out tonight and I will submit a pull request.

closed 2017-07-25 by BarbourSmith

krkeegan, 2017-07-26:
This is irritatingly not fixed.

Even with a slower feedrate I am still having issues.  What I am seeing looks like the PID process is fighting something.  I am seeing the zaxis movement become very jerky at times for no reason.  When it becomes jerky, it runs out of time to finish its full movement.  This is occuring on the downward as well as upward directions.

krkeegan, 2017-07-28:
OK pretty sure edadd40507309acefcbb0123c21c56c61f0941cf fixed this.

BarbourSmith, 2017-07-28:
Let's keep an eye on it and if it doesn't come back for a few days close the issue?

PR #267 (2017-07-25): Fix ZAxis Feedrate and Calculation of Total Distance
Fixed the calculation of moveDist which if the starting depth was negative and the finishing depth was positive was returning the wrong value.

Also discovered that the hardcoded feedrate for zaxis moves of 45mm/min was too much for my motor to keep up with.  I believe this was the main cause for the incomplete retraction before moving.

Now, the feedrate can be set by Gcode, but its upper bound is limited by the rpm of the motor.  With the stock motor this results in a feedrate of about 40mm/min.  But this number will adjust based on the pi [... cut]