MaslowCNC/Firmware #273: Error when trying to compile v0.82 through arduino app on OS X

designfrontier, 2017-07-27:
`Arduino: 1.6.8 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
sketch/number.c: In function 'bc_str2num':
sketch/number.c:1523:7: warning: assignment discards 'const' qualifier from pointer target type
   ptr = str;
       ^
sketch/number.c:1548:7: warning: assignment discards 'const' qualifier from pointer target type
   ptr = str;
       ^
In file included from /Users/designfrontier/projects/maslowcnc/Firmware/cnc_ctrl_v1/cnc_ctrl_v1.ino:15:0:
sketch/CNC_Functions.h: In function 'void executeGcodeLine(const String&)':
sketch/CNC_Functions.h:1189:19: warning: passing 'const String' as 'this' argument of 'String& String::operator=(const char*)' discards qualifiers [-fpermissive]
         gcodeLine = "";
                   ^
sketch/CNC_Functions.h: In function 'void interpretCommandString(const String&)':
sketch/CNC_Functions.h:1235:23: warning: passing 'const String' as 'this' argument of 'String& String::operator=(String&&)' discards qualifiers [-fpermissive]
             cmdString = cmdString.substring(secondG, cmdString.length());
                       ^
failed MSpanList_Insert 0x30e5a0 0xabc12febdb78 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2c6d60, 0x30e5a0)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2c6c20, 0x6, 0x10000000034, 0x10359)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2d0fd8, 0x2d21b8)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x307000, 0xc200000034, 0xc208012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc2080f0f18 sp=0xc2080f0f10
runtime.mallocgc(0x2200, 0x1a18c0, 0xc200000000, 0xc2081dc800)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc2080f0fc8 sp=0xc2080f0f18
runtime.newarray(0x1a18c0, 0x20, 0xaf9a)
	/usr/local/go/src/runtime/malloc.go:365 +0xc1 fp=0xc2080f1000 sp=0xc2080f0fc8
runtime.hashGrow(0x1b4c40, 0xc208155770)
	/usr/local/go/src/runtime/hashmap.go:744 +0x86 fp=0xc2080f1030 sp=0xc2080f1000
runtime.mapassign1(0x1b4c40, 0xc208155770, 0xc2080f1148, 0xc2080f1138)
	/usr/local/go/src/runtime/hashmap.go:456 +0x568 fp=0xc2080f10d0 sp=0xc2080f1030
arduino.cc/builder/props.PropertiesMap.Merge(0xc208155770, 0xc2080f11f8, 0x1, 0x1, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/props/properties.go:162 +0x153 fp=0xc2080f11c8 sp=0xc2080f10d0
arduino.cc/builder/props.PropertiesMap.Clone(0xc2080d2570, 0xc208154210)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/props/properties.go:170 +0xad fp=0xc2080f1220 sp=0xc2080f11c8
arduino.cc/builder/builder_utils.compileFileWithRecipe(0xc2080a4240, 0x5a, 0xc2080107e0, 0x68, 0xc208038300, 0x5f, 0xc2080d2570, 0xc2081f4e20, 0x2, 0x2, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder_utils/utils.go:131 +0x66 fp=0xc2080f1378 sp=0xc2080f1220
arduino.cc/builder/builder_utils.compileFilesWithRecipe(0xc20800c800, 0xe, 0x10, 0xc2080a4240, 0x5a, 0xc2081a0600, 0x11, 0x20, 0xc208038300, 0x5f, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder_utils/utils.go:120 +0x1c3 fp=0xc2080f14b0 sp=0xc2080f1378
arduino.cc/builder/builder_utils.compileFilesWithExtensionWithRecipe(0xc2081eed80, 0x8, 0x8, 0xc2080a4240, 0x5a, 0xc2080a4901, 0xc208038300, 0x5f, 0xc2080d2570, 0xc2081f4e20, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder_utils/utils.go:87 +0x248 fp=0xc2080f15b0 sp=0xc2080f14b0
arduino.cc/builder/builder_utils.CompileFiles(0xc2081eed80, 0x8, 0x8, 0xc2080a4240, 0x5a, 0x1, 0xc208038300, 0x5f, 0xc2080d2570, 0xc2081f4e20, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder_utils/utils.go:75 +0x4c7 fp=0xc2080f1698 sp=0xc2080f15b0
arduino.cc/builder/phases.compileCore(0xc208038300, 0x5f, 0xc2080d2570, 0x0, 0x7fff5fbffc8b, 0x4, 0x318f90, 0x2c0908, 0x0, 0x0, ...)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/phases/core_builder.go:86 +0x66a fp=0xc2080f1818 sp=0xc2080f1698
arduino.cc/builder/phases.(*CoreBuilder).Run(0x2c0908, 0xc20803c570, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/phases/core_builder.go:54 +0x410 fp=0xc2080f1978 sp=0xc2080f1818
arduino.cc/builder.runCommands(0xc20803c570, 0xc2080f1b90, 0x1d, 0x1d, 0x19501, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:187 +0x139 fp=0xc2080f1a08 sp=0xc2080f1978
arduino.cc/builder.(*Builder).Run(0xc2080f1d88, 0xc20803c570, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:118 +0xef7 fp=0xc2080f1d68 sp=0xc2080f1a08
arduino.cc/builder.RunBuilder(0xc20803c570, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:218 +0x49 fp=0xc2080f1d90 sp=0xc2080f1d68
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/main.go:320 +0x1eb2 fp=0xc2080f1f98 sp=0xc2080f1d90
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc2080f1fe0 sp=0xc2080f1f98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc2080f1fe8 sp=0xc2080f1fe0
arduino-builder returned 2
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

designfrontier, 2017-07-27:
and now it compiled... who knows....

closed 2017-07-27 by designfrontier

BarbourSmith, 2017-07-27:
Computer are WEIRD 

I'm glad it fixed itself... 😶 

If it seems like a recurring issue, or if you find out what was going on let me know so I can better help people in the future