MaslowCNC/Firmware #277: Closed loop breakdown needs to send alert back to ground control.

JDogHerman, 2017-07-28:
When a motor is disconnected while ground control is running the firmware does not stop the run.

If a break in the closed loop fails the cut should be halted and not continue to run.

BarbourSmith, 2017-07-28:
👍 I agree

I think the way to implement this is by watching both the applied voltage (`_pidOutput`) and the measured speed (`_currentSpeed`) variables in the MotorGearboxEncoder object

blurfl, 2017-07-29:
Would this check be called from the main loop, and post a ‘!’ stop if triggered?

BarbourSmith, 2017-07-29:
If we called it from the compute speed function it would be easier I think. I love the idea of posting a '!'. That's a great way to reuse our existing stop apparatus!

blurfl, 2017-08-04:
I’ve looked into watching _currentSpeed and _pidOutput, but haven’t figured out a reasonable value of _pidOutput to use as a trigger. Can you suggest a value?

davidelang, 2017-08-04:
On Fri, 4 Aug 2017, Scott Smith wrote:

> I’ve looked into watching _currentSpeed and _pidOutput, but haven’t figured out a reasonable value of _pidOutput to use as a trigger. Can you suggest a value?

Both motors hitting 100% for more than a few seconds??

blurfl, 2017-08-04:
Ok, thanks. I’ll give that a try.

BarbourSmith, 2017-08-04:
I've been imagining two criteria

1) The motor hitting 100% for a time period (1s?) ---> PID output is 255
2) The speed is zero                                               ----> PID input is close to 0

blurfl, 2017-08-04:
I found that triggering on PID output > 1.5 and speed 0, the last part of a z movement tripped the trap (sticky motor slide? ;) ) I didn’t look at _how_long_ it had been in that state, though. That might be the key. Is 5 seconds reasonable?

BarbourSmith, 2017-08-06:
I can see how we would need a time threshold too because the motor will momentarily be stopped right at the beginning and end of the move. I'd guess that 1 second or even 1/2 second should be enough. That seems like an eternity in for the microcontroller

blurfl, 2017-08-06:
I’ll play with some different values and report back 🙂

JDogHerman, 2017-08-25:
Just a follow up this appears to still exist on GC 0.87 and FW 0.87