MaslowCNC/Firmware #282: Stutter in left motor

designfrontier, 2017-08-01:
https://www.youtube.com/watch?v=OjlU1UJpnog

The left motor seems to stutter when cutting and moving z-index on firmware v0.82 for me. The video shows it moving in and out. The video is of it cutting a 1 1/8" circle in 3/4 ply.

Cut settings are 25 ipm feed rate, 30ipm z feed rate, 0.15 in step down. Wasn't as pronounced in v0.80 for me but now it is very pronounced. Even vertical cuts are showing a heavy stutter.

I'll see if I can pull the correct lines out of the gcode file when I have some time. The right motor remains steady and stutter free.

davidelang, 2017-08-01:
it seems to be problems with the PID tuning being overly aggressive, there are 
several threads on this, look at the forum topic "do I have a bad motor" (or 
something similar to that)

There is an experimental GC and firmware branch to allow tuning of these.

David Lang

On Tue, 1 Aug 2017, Daniel Sellers wrote:

> Date: Tue, 01 Aug 2017 12:41:05 +0000 (UTC)
> From: Daniel Sellers <notifications@github.com>
> Reply-To: MaslowCNC/Firmware
>     <reply+0021b24fd780567537b6523f8ce7006a02dd72443dc0499a92cf00000001159838e
>     192a169ce0eb9c487@reply.github.com>
> To: MaslowCNC/Firmware <Firmware@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [MaslowCNC/Firmware] Stutter in left motor (#282)
> 
> https://www.youtube.com/watch?v=OjlU1UJpnog
>
> The left motor seems to stutter when cutting and moving z-index on firmware v0.82 for me. The video shows it moving in and out. The video is of it cutting a 1 1/8" circle in 3/4 ply.
>
> Cut settings are 25 ipm feed rate, 30ipm z feed rate, 0.15 in step down. Wasn't as pronounced in v0.80 for me but now it is very pronounced. Even vertical cuts are showing a heavy stutter.
>
> I'll see if I can pull the correct lines out of the gcode file when I have some time. The right motor remains steady and stutter free.
>
>

designfrontier, 2017-08-01:
@davidelang is it the triangular kinematics branch?

davidelang, 2017-08-01:
no, that is for dealing with a sled with different chain mounting.
kk

BarbourSmith, 2017-08-01:
You want the master version of both the firmware and Ground Control. Both need to be updated to 0.83 (the version on master of both right now) for the changes to work

designfrontier, 2017-08-01:
thanks @BarbourSmith I just saw that in the forum. It will probably be tomorrow at the earliest, and the weekend at the latest, before I can test it out.

designfrontier, 2017-08-03:
did some test cuts this morning... much much smoother. A little chatter on z moves from the right motor. Is there a place in groundcontrol to tweak the PID settings? I could not for the life of me find it

BarbourSmith, 2017-08-03:
Yes! It's under Advanced Settings (open settings, then click the Maslow Settings at the top). I know it's a little hidden.

If you find values that seem better, let us know so we can test them out!

designfrontier, 2017-08-03:
Will do!

designfrontier, 2017-08-15:
I am going to close this. The main thing I've noticed is some real heat sensitivity with the shield. The longer runs in my un cooled garage produced gradually more chatter in the cut. Initial cuts were fantastic with 0.83.

closed 2017-08-15 by designfrontier