MaslowCNC/Firmware #286: Enhancement - digital control of AUX1-3

blurfl, 2017-08-09:
Enhancement
There has been discussion in the forum about controlling power to the router using some of the AUX gpios. For reasons, I'm hesitant to write the M3/M4/M5 spindle control gcodes for that. I've proposed two options:

- A pair of Maslow-specific Bxx commands that would set an AUX pin to OUTPUT and set it HIGH or LOW. A required parameter would indicate which AUX pin to control.
- Implement the M42 gcode used in the Marlin branch of grbl. M42 accepts a pin number and requires a speed parameter that it uses to set a PWM speed if it is able, or HIGH/LOW if not. M42 defaults to the LED_BUILTIN if the pin number is invalid or missing.

I have these ready as separate PRs, based on v0.83. I'll wait to see if there is further interest before opening a PR.

davidelang, 2017-08-09:
On Tue, 8 Aug 2017, Scott Smith wrote:

> For reasons, I'm hesitant to write the M3/M4/M5 spindle control gcodes for 
> that.

I really think we want the M3/M5 g-codes, otherwise they won't be used.

However, we do need maslow specific Bxx codes to set which AUX pin the M3/M5 
commands affect (and probably something similar to set which AUX pin the coolent 
command triggers)

And we also need some way to set the AUX pins manually

mrfugu, 2017-08-15:
Other Issues Related to G-Code documentation and implementation: Open: #3  #191  Closed: #122 
Note that #122 was closed w/o completion, as M3/M5 (and others) are included in GRBL implementation. 

Please document any G-Code additions in the [G-Code and MaslowCNC](https://github.com/MaslowCNC/GroundControl/wiki/G-Code-and-MaslowCNC) wiki page.

BarbourSmith, 2017-11-28:
Can we close this now that we've implemented the M family of codes? I know that the issue was opened to address different things, but there seems like a lot of overlap in functinality

mrfugu, 2017-11-29:
^ outside my pay grade, I was just collecting things together to make more sense of em.

closed 2017-12-05 by blurfl