MaslowCNC/Firmware #289: If GroundControl app is running a job and closes the motors continue to run.

JDogHerman, 2017-08-25:
X-Posted from GroundControl https://github.com/MaslowCNC/GroundControl/issues/401

blurfl, 2017-08-25:
GC could/should send the ‘Stop’ command in preparation to shutting down. Is that the idea?

skiz, 2017-11-18:
I found App.on_close, and gave this a try, however due to the threading model, it's going to take some work to actually send a reliable stop command before exiting.  Doing this from the arduino side with `IfSerial` watchdog may be a better solution.

krkeegan, 2017-11-28:
I am not sure the arduino side is the best solution.  From its end, all it can see is whether it is receiving data on the serial port or not.  A crash in groundcontrol and a hang on the computer would look the same.  

I don't seem them very often right now, but I know I have seen hiccups in the past where I have briefly lost a connection during a cut.  If we issued a stop command, then the job is lost, even though there is nothing wrong with the commands which were issued to the arduino.

I get the concern, I just don't know the best way to resolve them.  

Maybe the hold command instead?  But lines can be very long so this may not achieve what you want.

blurfl, 2017-11-29:
I agree with your concern. Let’s give it a pass.

closed 2017-11-29 by BarbourSmith