MaslowCNC/Firmware #301: Replace Hard coded Encodersteps Constant

krkeegan, 2017-09-13:
https://github.com/MaslowCNC/Firmware/blob/8ab71d8f757d964160b8d99a47b0218c38250766/cnc_ctrl_v1/MotorGearboxEncoder.cpp#L165

This constant 7364.0 line would cause errors for anyone using a non-maslow encoder.  It also isn't quite right as the value is actually 7363.77025037.

The constant should be calculated using the encoderSteps setting likely by querying its parent axis.

BarbourSmith, 2017-09-19:
Should be fixed in #306 

Great catch @krkeegan !

krkeegan, 2017-09-19:
Awesome, looks good to me.

designfrontier, 2017-09-20:
Do we want to convert to `7363.77025037` instead of `7364.0`? Not sure what the implications of that would be. Could go to a shorter form like `7363.8`

krkeegan, 2017-09-20:
@BarbourSmith 's solution is to calculate the constant using the encoder resolution:

https://github.com/MaslowCNC/Firmware/blob/4a9085ba8d4a7c09ad65af6161d3e977ddf28380/cnc_ctrl_v1/MotorGearboxEncoder.cpp#L161

I don't know how precision works in C but I assume that gets us the full precision.

closed 2017-09-21 by BarbourSmith