MaslowCNC/Firmware #302: Is it Appropriate to Calculate PID When Axes are Detached?

krkeegan, 2017-09-14:
I have no experience with PID calculations, but is it normal to continue calculating the PID values when the controlled device is not active?

The P and the D terms don't look like they would be affected.  But the integral term continues to increase/decrease until it reaches min/max. I suspect that this causes a bit of a jump whenever the axes are reattached after being detached.

We don't generally detach the axes in the middle of a cut, so this likely doesn't arise much.  I can think of a few cases:

1. Likely mostly on the very first move.
2. I think if a zaxis move takes more than 2 seconds, there is the potential that the left and right axes could be detached briefly.  
3. I _think_ if there were a sequence of zaxis moves, such as a plunge and then retract, that the left and right axes could be disabled for the entire axis retract time.
4. Finally, whenever we click hold.

https://github.com/MaslowCNC/Firmware/blob/7cbfecb67fa7288fa00368b069425aba0000a289/cnc_ctrl_v1/cnc_ctrl_v1.ino#L42-L43

https://github.com/MaslowCNC/Firmware/blob/7cbfecb67fa7288fa00368b069425aba0000a289/cnc_ctrl_v1/Axis.cpp#L99-L115

https://github.com/MaslowCNC/Firmware/blob/7cbfecb67fa7288fa00368b069425aba0000a289/cnc_ctrl_v1/PID_v1.cpp#L51-67

davidelang, 2017-09-14:
It is normal, but mostly because external forces could move the motors. In our 
case that will not happen for the main motors (thanks to the worm gear) but 
could happen with the Z axis.

I don't think that detaching/attaching is the answer, I think it's more a matter 
of "if the desired speed is zero, shut down power and stop calculating" or some 
other limit on power when the desired speed is zero

krkeegan, 2017-09-14:
When the firmware detaches (just the name of the function, it does the power down like you say) am axis, it can no longer control it.  So while you may be able to move the zaxis manually, the firmware cannot because the pin is disabled.  We are calculating the PID routine even though the output from it is ignored.

BarbourSmith, 2017-09-14:
I agree.

It sounds like we have two options, either we can not do the PID calculations when the axis is detached (easy), or we can have the controler well enough tuned that the motors will automatically power down (because PID output is zero) when they are where they should be (this is much harder without adding a large dead band which would hurt accuracy).

Let's start with option one, and work towards option two?

krkeegan, 2017-09-14:
I agree with that plan.

I actually started doing some PID tuning last night.  I was able to generate data to create pretty PID graphs.  I spent the night working on the velocity controller first (hence all my issues regarding the control loop).  I certainly made the graph look better, I think I can hear a subtle difference in the motors while doing normal movement.  The process was relatively easy, however I suspect that the positional controller will be much harder to tune.

BarbourSmith, 2017-09-14:
👍 👍 ❤️ 👍 

If you find values that are better than the defaults, or think any of the code to generate pretty PID tuning graphs is worth merging please PR it! I kinda liked using a "bullet" shaped object (a rectangle with one end rounded) for tuning because sometimes I found tunings that would handle smooth curves really well, but overshoot on the right angle or vice versa.

krkeegan, 2017-09-14:
So far I am just dumping the error values and plotting them in google sheets.  I will certainly submit the code, but I may not make a nice front end like the simulation page.  Ideally that would be the goal to make an interface where users could test various values, I am not sure I have the desire to construct all of that in kivy.

krkeegan, 2017-09-14:
It may be worth upgrading the PID library to 1.2.1 in order to take advantage of what Brett calls proportional on measurements:

http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/

It may be appropriate for the position controller.

BarbourSmith, 2017-09-14:
@krkeegan Those changes to the PID controller sound HUGE. That sounds like it might be exactly what we need. FANTASTIC FIND! 👍 👍 👍 💐 👍 . I fully support switching to the PonM system right away, or at least exploring it. Can anyone see any drawbacks? Does does anyone have experience (beyond the article) with this kind of system?

I agree that we don't need to do plotting in Kivy, especially since Kivy doesn't lend itself to making plots very easily

PR #309 (2017-09-24): Don't Run PID Calculation if Motors are Detached/Unpowered
This was causing the integrator term to wind up while the machine was on, but not running.

Fixes #302
closed 2017-09-24 by BarbourSmith