MaslowCNC/Firmware #303: Save Integral Term to EEPROM

krkeegan, 2017-09-14:
When the Arduino starts up the integral term is zero.  As a result the first few moves are less than accurate until enough "history" had built up in the term to be of use.  It may not be a big deal, but it would be worthwhile to save the term to EEPROM to avoid this.

BarbourSmith, 2017-09-14:
Great suggestion! I know exactly what you are talking about, but I hadn't thought of a solution

krkeegan, 2017-09-27:
At the moment, this isn't an issue because the newer suggested PID tunings don't use Ki values at all.

BarbourSmith, 2017-09-27:
I would feel good about closing this one after merging 😀

krkeegan, 2017-11-28:
Closing, as I don't think using Ki will ever be a good idea without some substantial work.

closed 2017-11-28 by krkeegan