MaslowCNC/Firmware #310: File from DXF2Gcode does not run correctly

BarbourSmith, 2017-09-24:
This file generated from DXF2Gcode does not appear to run correctly: 
[test_shapes2.zip](https://github.com/MaslowCNC/Firmware/files/1326956/test_shapes2.zip)

Parts of the circle and the z-axis commands are ignored. The original discussion of the issue can be found here: http://forums.maslowcnc.com/t/circles-and-z-axis-problem/713

GeroBH, 2017-10-15:
Would need someone to confirm this finding:
In the test file the sled teleports from X  35.418 Y  55.809 to X  63.702 Y  84.094.
If I replace the double blank spaces in line 18 with a single blank space, the teleportation does not take place. So instead of (blank is symbolized as ^):
G2 X^^63.702 Y^^84.094 I^^14.142 J^^14.142
try
G2 X^63.702 Y^84.094 I^14.142 J^14.142
CAMotics simulates the circle cut well double blanks.

blurfl, 2017-10-16:
This looks like it can be fixed with a single line addition to GroundControl. [PR#434](https://github.com/MaslowCNC/GroundControl/pull/434) is the change. @GeroBH, would you have a chance to try this to see if it solves the issue?

GeroBH, 2017-10-16:
It solves the issue with the file location not stored and the teleportation of the sled. Great Job.
(Not sure this was the correct way, I renamed the original gcodeCanvas.py to gcodeCanvas-org.py, took the raw text of the commit and pasted that into a text file saved as gcodeCanvas.py in the UIElements folder of GC. V0.94

BarbourSmith, 2017-10-16:
Excellent work @blurfl, the changes are merged now so downloading from master will fix the issue too!

blurfl, 2017-10-16:
@GeroBH gets the credit here, he had the patience to find the cause.

GeroBH, 2017-11-28:
I think this can be closed :-)

BarbourSmith, 2017-11-28:
Great work everyone!

closed 2017-11-28 by BarbourSmith