MaslowCNC/Firmware #323: Chains extend too short

BarbourSmith, 2017-10-26:
From the original forum post [here](https://forums.maslowcnc.com/t/calibration-how-to/806/6):

> Hello, I am new to this forum, and writing from France. Congratulations for the forum, it is very helpful! Unfortunately I have been trying the calibration processes, both automatical and manual, without success (temporary sled so far).
My frame is larger than the standard one, because I had a nice 30mm thick, 3100 x 1530mm sheet of plywood. It is:
Distance between motors: 3666mm
Work Area width: 3100mm
Work area height: 1530mm
Motor Offset height: 345mm
Distance between sled mounting points: 211mm
vertical distance sled mounts to cutter: 140mm

> When the chains finish being automatically extended, they are too short, I can’t reach the sled, something like 400mm or so is missing.
Arduino 1.8.5, Maslow Firmware Version 0.94, ground control 0.94

> Would someone be able to help me? Thanks!

The issue seems to be that the calibration process extends a fixed length of chain. The correct behavior (I think) would be to extend enough chain to reach the middle. At this point in the calibration process we have rough estimates for all of the measurements so there is no reason we shouldn't try to center sled, right?

davidelang, 2017-10-26:
On Thu, 26 Oct 2017, BarbourSmith wrote:

> The issue seems to be that the calibration process extends a fixed length of
> chain. The correct behavior (I think) would be to extend enough chain to reach
> the middle. At this point in the [...]

Agree, this step in the calibration is assuming that you are using a stock size 
machine, if you have different dimensions, this won't work.

At this point in the calibration, we know the distance between the motors, we 
should be able to calculate the max height that the sled should be able to go 
based on this (33 pounds of force from each motor, what is the highest angle 
that we can do), extend the chains out until they meet at this point. This will 
put the sled down from that point a bit (how much depends on the motor mounts).

We should also allow the user to feed out additional chain at this point if they 
want to move the sled down. We will still know how much chain has been fed out.

I would strongly consider finding the center at this point in the process, feed 
out enough chain for the user to attach the sled to the chain, then have the 
user move the sled up or down until it's at the center of the work area. This 
will stop if from doing wildly wrong things if someone is doing a 8x4 machine of 
2x4 machine instead of the stock 4x8 machine.

see the discussion in the triangular calibration for more on this topic

blurfl, 2017-10-26:
Let’s be careful with this. Extending the chain this fixed amount makes the manual calibration  using the painted link possible regardless of changes in motor placement and sled size. 
I would suggest that we put a return to center button in the calibration dialogue. That would handle the issue brought this up,  and preserve the present manual calibration function. 
Further, the calibration dialogue could report the length of chain needed for the machine geometry. If the user needed more chain, it would alert them to this.

BarbourSmith, 2017-10-26:
Great point @blurfl I knew I was forgetting something. I remembered thinking about this before and deciding not to do it for some reason. Thanks for the second set of eyes. 

What if the automatic calibration process simply ran a "return to center" command automatically after both chains were calibrated? I guess that would mess with the user's ability to mark the chains in the first place. 

What if the sled would "return to center" when the 'finished' button in that step is pressed? That way it would be both automatic (I feel like the more options we give everyone, the more mistakes will happen), and it would still support the manual calibration mode

blurfl, 2017-10-26:
I like the 'Return to Center' on 'finished'. Let's make sure that the user knows that the motors will come alive and the sled will move, though... what do you think about alerting if additional chain will be needed?

BarbourSmith, 2017-10-26:
I think alerting if additional chain will be needed is a great idea. 

We can trigger an alert popup when that step is entered if more chain well be needed?

blurfl, 2017-10-26:
A pop up, or maybe just text in the window saying ‘ the stock chain is 10 feet long,  that should be enough to reach all points on your work area ‘ or ‘ The stock chain is 10 feet long, it will take 15 feet of chain to reach all points on your work area ‘

davidelang, 2017-10-27:
you should be painting the chain after they are run out the proper amount.

Currently, the 'proper amount' is a fixed amount that only works if you have a 
stock size frame. Changing it so that the 'proper amount' is proper for that 
machine shouldn't cause any problems.

davidelang, 2017-10-27:
On Thu, 26 Oct 2017, BarbourSmith wrote:

> Great point @blurfl I knew I was forgetting something. I remembered thinking about this before and deciding not to do it for some reason. Thanks for the second set of eyes.
>
> What if the automatic [...]

'return to center' may tear the sled apart (if the motors are 12' apart) of drop 
it to the ground (if the motors are 4' apart) if it means moving to the default 
chain lengths.

the current definition of center is bogus for non-standard machines.

blurfl, 2017-10-27:
Painting the king link should be independent of the sled or the machine dimensions. That way it doesn't change when one changes sleds or re-runs the motor distance measurement.Painting the king link allows one to bypass the tedious step of paying out chain during the process of putting the machine into a known state. 
 'Return to Center' is where the sled and machine geometries enter, the next step in the process. We've just skipped a great portion of the chain movement, so this step is quicker. It is a movement to the point in the geographic center of the workarea. 
 The 'Return To Center' won't rip a sled apart or drop it to the ground unless the workarea was incorrectly specified. The chain movement of the 'Automatic Chain Calibration' can't, since it starts with the chain ends on the sprockets. In the situation that started this issue, the 'Return...' command made the chains long enough to hang the sled on a very large frame. On a small frame, it would reel chain back in. 
 The current definition of center uses the values measured by calibration of the sled and the machine to calculate it, in Kinematics::recomputeGeometry(). It really has nothing to do with the chain movement in the 'Calibrate Chain...', which is only to put the chains into a known configuration.
 Maybe the confusion here is the difference between these:
1) chain calibration, which is used to mean putting the machine into a state with a known amount of chain payed out
and
2) Return to Center, which moves the sled to the center of the workarea as calculated from the values measured and/or entered by the user in the calibration process.
and
3) Home, which is a setting controlled by the user, both with the on-screen buttons or with the Settings dialog where he can choose the default location such as (0,0) that GC uses when it starts up.

BarbourSmith, 2017-11-01:
Hopefully fixed in #324 and GC #422

The chains will now extend to the known distance as before during the calibration step, then the sled will be moved to the center of the sheet when the finish button is pressed for that step.

closed 2017-11-01 by BarbourSmith