MaslowCNC/Firmware #340: Tool Change Does Not Give Pause for T0

madgrizzle, 2017-12-12:
Tool change routine in CNC_Functions.h only pauses if the tool is changed to something other than 0.  So trying to change back to Tool 0 is ignored.  Perhaps should be 

            if (nextTool != lastTool) {
                setSpindlePower(false); // first, turn off spindle
                Serial.print(F("Tool Change: Please insert tool "));   // prompt user to change tool
                Serial.println(nextTool);
                lastTool = nextTool;
                pause();
            }

BarbourSmith, 2017-12-12:
I agree, that seems like a good solution. We want to make sure it doesn't pause for the first tool (because 90% of the time there is only one tool in the program so no need to pause), but we want it to pause for every tool after that even if it is 0 again

madgrizzle, 2017-12-12:
I see the desire, but you may need to send a special character to denote start of file.  I don't see something like that in the .nc files makercam has produced for me.

krkeegan, 2017-12-12:
Well, if the default value of lastTool matches the default tool number used by your CAM software, I think it would skip the tool change on startup then.

davidelang, 2017-12-12:
I think it would be good to prompt even on the first change of a file if it's 
different from the last one used. The user may have changed the too, and if so, 
it's a trivial step to get through, but if they forgot, it's a good safety net

madgrizzle, 2017-12-17:
There's apparently two places where tool changes get processed.   As 1.00 stands, the line
` if ((tNumber > 0) && (tNumber != sys.lastTool))` in executeOtherCodeLine is causing the issue.

davidelang, 2017-12-17:
are you running 0.99 or the latest from github? I thought I saw changes for this 
go in in the last week (which would not be in a release yet)

madgrizzle, 2017-12-17:
No, I'm running 1.00 (GC and FW).  There was a change elsewhere in the code, but apparently there are two places that it needed to be corrected.

madgrizzle, 2017-12-17:
Ok, found the issue with my my tool change didn't work.  My gcode sent the following sequence of tool changes:  T1, T1, and T0.  The first T1 resulted in a pause, but when it did the second T1, nextTool was set to 0 (my comment above).  So when the T0 was sent, the firmware already thought T0 was installed and so it didn't pause.  I'll make a PR for people to consider.

BarbourSmith, 2017-12-18:
Excellent catch!

krkeegan, 2017-12-27:
I think this is fixed by @madgrizzle commit.  Let me know if this isn't the case and we can reopen this.

closed 2017-12-27 by krkeegan