MaslowCNC/Firmware #349: Pointer warning

davekopecek, 2017-12-17:
I'm getting the following warning when I upload 0.99 to the Maslow
 
````
sketch\number.c: In function 'bc_str2num':
sketch\number.c:1523:7: warning: assignment discards 'const' qualifier from pointer target type
   ptr = str;
       ^
````

blurfl, 2017-12-17:
The warning is a nuisance, but it can be ignored. Here's a previousl [discussion.](https://forums.maslowcnc.com/t/errors-shown-while-compiling-and-uploading-arduino-code-to-arduino-boa/556])

BarbourSmith, 2017-12-18:
Good to close?

krkeegan, 2017-12-18:
Yeah, I think I have tried to squelch this error before.  Are we even using the number library anymore?  It was also part of bcconfig.h?  Anyone remember why these were included, I don't think we use them anymore and dropping them from the repo would get rid of this warning.

BarbourSmith, 2017-12-18:
No, you are totally right I think we can safely remove them. They were used at one point in the kinematics to address rounding errors, but that issue was fixed quite a while ago I believe

closed 2017-12-27 by BarbourSmith