MaslowCNC/Firmware #352: Add Calibration Functions

krkeegan, 2017-12-27:
@davidelang described it well

I'm thinking that we need just a couple extra commands

I believe we already have commands in place to set the position of a motor to X
and to move the motor (absolute movement, relative movement, apply X% power in a
direction), so I believe we only need a couple extra commands

1. use the current motor position to define the distance between the motors

   Assumes that you zero'd both motors, ran chain out from one motor, hooked it
over the other motor and pulled the chain tight before executing this command.

2. triangular calibration

   As I tried to specify this, I realized it will need to be two commands

2a. Triangular calibration step 1 "I am at Y=x" (defaulting to 0)

   Causes the firmware to remember the chain lengths for this coordinate

2b. Triangular calibration step 2 "I am at Y=x2"

   The firmware knows where it thinks it is, and what the chain lengths are for
here. So given the actual position, and the memory from step 2a, it can
calculate the chain offset and y offset values.

The process of using these would be to feed out chain from both sides to the
same amount, drive the sled to the first location, issue command 2a. drive the
sled to a second position (either a pre-marked position or to an expected
position and then mark it, move the sled, measure it, move the sled back) and
issue command 2b.

https://forums.maslowcnc.com/t/calibration-pattern-for-triangular-kinematics/879/61

I go into a way to do calibration for triangular kinematics from a pure math
point of view, @rjon17469 implements something similar in excel doing some
brute-force iteration to find the results.

Basically you use the fact that you know:

0. the math is much simpler if you are exactly at the center of the machine
1. the distance between the motors
2. the chain length at one position
3. the chain length at a second position
4. the distance between these two positions

from this you can calculate the 'missing' chain length and the Y offset needed
to make things work.

It works best if you can use the exact same code for the calibration iterations
that you do for the actual movement (no chance of the code getting out of sync
between the two), which makes doing this in firmware ideal.

davidelang, 2017-12-27:
for the motor distance calculation.

the motors are moved to get one tooth straight up, and position is set to 0 for that motor.

movement happens on both motors (left motor in the + direction, right motor in the - direction) to feed out chain and pull it tight.

while it is tight, issue the 'calculate motor distance' command

This will do the calculation:
<distance between motors> = (<left motor encoder steps> + <right motor encoder steps>) * <chain pitch> * <tooth count> / <encoder steps per revolution>
and save this value in eeprom (as well as sending it as a message to GC)

davidelang, 2017-12-27:
note that the + between the motor steps is because the right motor will have moved in the negative direction to take up the slack, so the calculation is actually "number of steps the left motor has move clockwise - number of steps the right motor has moved clockwise"