MaslowCNC/Firmware #356: Should we have an Exit Command and Check on Reboot?

krkeegan, 2017-12-27:
The encoder steps are saved to eeprom as soon as all three axes are idle.  This can only happen if the Arduino is still powered.  I think this is generally true whenever the Arduino is plugged into the USB?

I think there exist edge cases where a user shuts down GC quickly, or where something else happens and the steps are not saved to eeprom, but the user isn't warned of lost position data.  I don't have a good feel for how often or if this happens.

BarbourSmith, 2017-12-27:
Yes! I think this is a great idea. We try to add similar behavior now where if the connection is lost while a gcode program is running GC will display a popup saying that calibration has likely been lost, but this would catch not only that case, but also others

davidelang, 2017-12-27:
On Wed, 27 Dec 2017, Kevin Robert Keegan wrote:

> The encoder steps are saved to eeprom as soon as all three axes are idle. 
> This can only happen if the Arduino is still powered.  I think this is 
> generally true whenever the Arduino is plugged into the USB?

the arduino gets it's power from the motor controller board, so if you hit an 
e-stop type power switch, it has no warning before loosing power.

> I think there exist edge cases where a user shuts down GC quickly, or where 
> something else happens and the steps are not saved to eeprom, but the user 
> isn't warned of lost position data.  I don't have a good feel for how often or 
> if this happens.

it probably happens a lot in the early stages as people are doing things that 
causes the machine to 'go crazy', and then as they learn the machine and how to 
use it, the frequency of it happening drops drastically.

davidelang, 2017-12-28:
when the maslow starts moving, have it set a 'moving' bit, and when it saves the position (from all three axis being idle), it should clear this bit.

The one thing that bothers me about this is when a user is jogging around (moving the sled manually), that results in lots of starts and stops, and therefor a lot of writes to the eeprom.

how about something along the lines of:

if we get just a single line of g-code, execute it but hold off on changing the position for a few seconds.

if we get multiple lines of g-code, set the 'saved position invalid' bit and save the position (clearing the bit) as soon as all three axis are idle and there is no more serial data being processed.

thoughts??

blurfl, 2017-12-28:
The Arduino gets it’s power from the USB connection. If you interrupt power from the motor power supply, the Arduino waits for movement to restart on reconnection of motor power, and continues with the cut.
If you pull out the USB cable, or the laptop goes into a sleep that interrupts USB, the Arduino would be affected.

krkeegan, 2017-12-28:
> when the maslow starts moving, have it set a 'moving' bit, and when it saves the position (from all three axis being idle), it should clear this bit.
>
> The one thing that bothers me about this is [...]

I think this is probably the best idea.  When movement is started I would enable the bit in eeprom.  Then whenever we write the final steps I would clear the bit at the same time.

The number of writes would be 2x the number of writes we make for the step counts.  Clearly more, but I don't think it is in the range of a problem for the 100,000 cycles.   

My opinion would be at $40, I think having to replace the arduino is better than wasting a 4x8 sheet of plywood because your machine became subtly un-calibrated.

krkeegan, 2017-12-28:
> If you interrupt power from the motor power supply, the Arduino waits for movement to restart on reconnection of motor power, and continues with the cut.

I know people have talked about using this as an e-stop.  Does anyone know, do we continue to get step counts from the motor shield even when it is unpowered?  If we don't. that is an issue and people need to be recalibrating after using that as an e-stop.

BarbourSmith, 2017-12-28:
I agree.

The shield will still read the encoders even if it is not powered so we're OK on that front.

I think the 100k cycles on the arduino is probably pretty conservative. Someone wrote a program to test how long it really takes to fail and they went over a million:

https://www.youtube.com/watch?v=t8X0zip7TvU

krkeegan, 2018-02-24:
This should be solved by bfab8ea3becfd81e54f8f98296bdc23a3d0761fc, just need to test a bit.

krkeegan, 2018-02-27:
So, I think this has to wait until we have a functional alarm structure in the code.  That has to wait until we replace the B code commands (so that we can more easily classify commands which are exempt from the alarm state).