MaslowCNC/Firmware #357: How to Abort When Calibration Settings are Wrong

krkeegan, 2017-12-28:
Maybe this isn't an issue or maybe this should be solved in another way.

We had a few issues recently where the rotation radius was entered incorrectly.  The result is that when the machine is commanded to move it moves in a single direction (no matter which direction is commanded) at full speed.  This can be a bit frightening and potentially dangerous.  

I was trying to think of a way to catch this issue. A check of the calibration numbers in GC catches the egregiously wrong numbers.

Would tripping a fault state if the position error of an axis exceeded some threshold make sense? The position error in mm = the current chain length - the desired chain length.  In normal operation this number less than 1.  

Setting the threshold to something like 40mm should provide more than enough buffer.  I can't think of a situation in which a user should ever be in this range.  This would also have the added benefit of halting the machine if it is unable to keep up with the feedrate as well.

This wouldn't save us in all cases.  If the sled is near the center top, an error in both chains of less than 40mm could still destroy the linkage arms.

Thoughts?

BarbourSmith, 2017-12-28:
I think this is a good idea. The one caveat I would add would be to make it easy for someone down the road to change the definition of "acceptable" dimensions easily maybe with a #define. I know people build their frames all different sizes so what seems out of of wack for one person might be right for another

davidelang, 2017-12-28:
I was thinking along similar lines for other reasons.

When the position is being reported to GC via serial, it should include the 
known error.

If this error is too large, we should stop. How large should be a configurable 
item.

In the short term, we will have some error as we start and stop since physics 
don't allow for instant starts and stops.

How large can this error be today? (as we implement acceleration, it will be 
smaller)

one thing we may need to do is at the end of calibration, recalculate where we 
currently are based on the current chain lengths and settings. If that isn't 
done, it could throw things off.

halting if we are too far from where we should be is a good idea, that should 
never happen.

David Lang

krkeegan, 2017-12-29:
> one thing we may need to do is at the end of calibration, recalculate where we
> currently are based on the current chain lengths and settings.

We do this, but I don't think there is any check that the position is in fact on the work surface.  Or even realistic.  

So maybe I hold off on the positional error idea.  Perhaps putting the machine into an alarm state if the forward calculation places the sled off the worksurface will solve our issue.

blurfl, 2018-01-10:
This sounds like an enhancement, yes?

krkeegan, 2018-01-10:
Yes, I assigned it to myself, I have some work done on this as part of a larger alarm-locking mechanism.