MaslowCNC/Firmware #358: Jerky motor movements after clicking the stop button

BarbourSmith, 2017-12-28:
An issue related to jerky motor movements after clicking the "stop" button has been reported in the forums. The full thread can be found [here](https://forums.maslowcnc.com/t/stop-on-ground-control/521/8)

The gist of the issue is that: 

"I wanted to add to this topic. I’ve been playing with my kit , learning the software and also noticed funky actions with the STOP button. When I click it, it stops the motors. But if I then go try and manually move , the motors are very jerky and sound like they are binding up. The fix is obviously to restart Ground Control. But I just wanted this pointed out in case moving after a STOP might be hurting the motors or the maslow shield drivers. Using 1.0

Here is a video. So I clicked move 8 inches downward at the start , then clicked the STOP button about 4 seconds into video. Then you can see how the sound and movement changes when I click move 8 inches again.

BTW, it took me two tries this time at clicking stop during a move, to get it to sounds (different)

https://www.youtube.com/watch?v=Rfj-v4NZ3YQ

Both motors definitely exhibit a slower jittery sounding movement compared to the faster smooth movement prior to the STOP.
Strangely, I couldnt get the issue to revert even by restarting Ground Control, repowering arduino, even trying 0.99 firmware.
I then went into actions and went thru the first step in calibrating machine, backed out of that. Then went into calibrate chain length, which I did nothing but just click through. At end, it leaves one motor running (still noisy). So I closed Ground Control, reopened which stops that motor. Then Magically, the motors run smooth again.

blurfl, 2018-01-10:
Has this been addressed by subsequent releases?

krkeegan, 2018-01-10:
I haven't had anytime to try and reproduce the issue to see if I see it as well.  I don't think I have changed anything specifically that would solve this.