MaslowCNC/Firmware #359: List of Settings Changes

krkeegan, 2017-12-29:
This is a list of changes I intend to make to the settings struct saved on the eeprom.  Since changes require that users re-push the settings to the device, I would like to bundle as many changes together

- Add Exit command as described in #356 
- Add error threshold as described in #357 
- Remove OriginalChainLength (this can be set and sent from GC)

BarbourSmith, 2017-12-29:
These look like excellent changes to me!

blurfl, 2017-12-29:
Makes good sense to bundle them and include them with the upcoming release.

blurfl, 2018-01-10:
Have these been done? Would an 'assignment' be appropriate? Also, as not a 'bug' is it an 'enhancement'?

krkeegan, 2018-01-10:
Umm, this one is a bit of a special case.  I envisioned this as a todo list that would aggregate all of the anticipated changes, then I would do them all at once and close this issue.  Starting a new one when the next changes popped up.  

I assigned it to myself.  I am happy to move the list off github if it bothers people, I just figured this would provide some notice of anticipated changes and allow people to object if I was altering a setting in a way they disagreed with.

blurfl, 2018-01-10:
It's a great list, please don;t move it. If you assign it to yourself it's clear that it's being addressed.

BarbourSmith, 2018-01-10:
Yes! I think this is a great thing. Having it here gives us all a chance to keep up with what the plan is and what needs to happen at the same time

davidelang, 2018-01-10:
it's far better for a list like this (and the discussions on any of them) to be 
here on github that to take it to the forums.

If we had a mailing list, it may be reasonable to keep it there, but while the 
forum can somewhat be treated as a list, the issue of searching for things makes 
it not a good replacement for this sort of thing.