MaslowCNC/Firmware #360: calibration problem switching from quad to triangular kinematics

davidelang, 2018-01-02:
in https://forums.maslowcnc.com/t/is-there-a-kinematics-calibration-guide/1702 we just had another case where switching kinematics models doesn't work as long as the settings for the quad kinematics are non-zero.

Those variables should not be used at all inside the triangular calculations, but we've now has numerous cases where they needed to bet set to 0 or the .ini file recreated from scratch before triangular kinematics work

since calibration is mostly a GC item, it's probably a bug in GC, but we should check the firmware just to be sure that the variables aren't used there

BarbourSmith, 2018-01-02:
Thank you for making an issue for this @davidelang .  I agree that this is an important issue.

krkeegan, 2018-01-02:
I will double check the firmware, but I am nearly 100% certain this can't be an issue in it.  I suspect this is an error in the calculation of the rotation radius in GC.

BarbourSmith, 2018-01-02:
Is there any chance we could be failing to switch the kinematics type and someone could be using the quadrilateral kinematics math with the triangular kinematics hardware? Then setting the distances to zero would result in an improvement

krkeegan, 2018-01-02:
I don't think so.  I can check tonight.  But I am pretty sure if you change from triangular to quadrilateral and change nothing else the reported position of the sled is dramatically different.  Which I think means this is working.

davidelang, 2018-01-02:
we've had a lot of people get nonsense results from the triangular calibration 
that have been solved after setting the quad related values to 0 or deleting the 
.ini file

a couple of them could be operator error, but this is getting to be a trend.

krkeegan, 2018-01-02:
I agree there seems to be an issue.  I just don't have enough data to determine what is causing it and I am doubtful that it is in the firmware.

davidelang, 2018-01-02:
I am doubtful about the firmware as well, but I filed a similar ticket in both 
places just in case.

remember, the symptom of the problem happening is that post-calibration, any 
movement just drops the sled to the floor.

It's possible that the machine dimensions can be off badly enough to cause the 
firmware to do this, but this seems like too consistant a problem to be sure of 
that without checking.

blurfl, 2018-01-10:
This seems to have been corrected when the user applied the default settings. I hesitate to label it a 'Bug' until it is reproducible.

BarbourSmith, 2018-01-10:
Great point @blurfl 

Should we have a tag for issues which we are unable to consistently reproduce? It seems like we run into a fairly high number of strange cases where something seems like a bug at first, but then goes away on it's own. We want to keep track of those issues to see if a pattern emerges, but we can close them if they don't show up again

blurfl, 2018-01-10:
A good point. Also might need a 'to do' label for items like #359?

davidelang, 2018-01-10:
something to indicate "yes, we agree that we want to do this, but haven't gotten 
around to it" is a good idea.

That could be the meaning of 'to do'