MaslowCNC/Firmware #361: Add Fake Servo Option for Testing

krkeegan, 2018-01-02:
It would be nice if there was an ability to test changes to GC and the Firmware on only an arduino. 

I think the biggest issue is accurately triggering the encoder steps.

I see two options:
1. Software based - add a precompile define that alters the code at the lowest level we can to trick it into thinking that the motors are turning at the proper speed.  I think this would have to be done in MotorGearboxEncoder::computeSpeed().

2. Hardware version - Trigger the encoder pins.  Is there some way to jumper a few pins on the arduino and trigger the encoders using other pins?  Alternatively we could have a second arduino with a simple sketch that looks at the pwm signal and outputs appropriate encoder signals.

I am leaning towards the software version unless we can figure out how to do the hardware with only a single arduino.

I am open to any other suggestions/ideas.

krkeegan, 2018-01-02:
I think I am going to do the software version.  I think there is a chance we could get a hardware version working using one of the other timers, but it adds complexity that could itself be brittle and you really don't want your testing interface to have its own issues.

idoc, 2018-01-02:
Neat idea!

I've been looking at a few things I might be able to help with (I'm especially interested in the work around GRBL compatibility and getting Maslow to work with controlling software other than GC), and it would be great to be able to test code and play around without having to hang out in the cold basement where my Maslow lives :)

(also, hi, everyone! I'm new here)

blurfl, 2018-01-02:
Hardware version - I'm using a pair of motors and PowerControl board with a power supply from the spare parts bin. Not the cheapest solution nor the most elegant, though.

GeroBH, 2018-01-03:
This is a great idea! I have this setup (no motor shield) at my home office almost since day one and it helped a lot of times.

I could replicate buffer overflows, not loading .nc , to large gcode files and other errors not related to actual feed back.

The only flaw I saw so far is that the position indicators are not matching positions. Surprisingly the red one still moves around.

Only if I doubt what I see, I go to the workshop and fire up the real thing.

 

Do we have 2 free pins on the Mega? If so, a check for a bridge could divert the software to the simulation part. Or better 1 pin pulled low to ground.

davekopecek, 2018-01-03:
It seems like it should be possible to spin up and arduino and ground control in docker containers for development. A quick google search shows that several people are working with containerized arduinos.   This would also nudge us a bit toward automated testing.

krkeegan, 2018-01-03:
A virtualized arduino would be cool if that is what docker can achieve.  I know nothing about docker so I would need a lot of handholding to get that going.

I found a cheap $13 arduino clone that appears to work well (although it has mixed reviews about the header pins being misaligned, which is fine for my use).  This seems like a cheap simple solution.

> Do we have 2 free pins on the Mega? If so, a check for a bridge could divert the software to the simulation part. Or better 1 pin pulled low to ground.

This is an interesting idea.  It should be easy to implement and then you wouldn't have to re-upload a changed firmware if you wanted to test the same thing live.  I will look into this.

krkeegan, 2018-01-03:
A friend suggested looking into an avr simulator.  This looks promising

https://github.com/buserror/simavr

We wouldn't have to change our code at all.  All we would need a simulation of the servos, which should be easy.  Sadly this doesn't run on windows only OSX or Linux.

BarbourSmith, 2018-01-03:
Is there any value to using it for integrated testing using something like CircleCI? I know its free for open source projects and the box they give you is running linux. I think we would benefit a lot from having automated testing of pull requests even if it was just running through sending each command and 'clicking' each button

krkeegan, 2018-01-03:
It might be worth looking into.  I think I have come back around to the easy option of doing testing on an arduino.  There seem to be a lot of comments about quirks and issues that arise in any virtualized implementation and arduinos are cheap in comparison to the time to set all of this up.

krkeegan, 2018-01-06:
I am opening a separate issue for simavr.  I think it would be worthwhile to accomplish this, but I don't have the time to achieve that right now.

In the interim, I found some cheap $13 Mega2560 on amazon with prime, for those others who also have an amazon addiction. 

http://amzn.to/2EbKJod  

Some reviews suggested to pinout spacing maybe wrong, but if it is, so is the one in the original kit because these are an identical match.

PR #362 (2018-01-08): Add a Simple Fake Servo Option
This is very basic.

If FAKE_SERVO is defined, each PID cycle it will look at what the last requested PID rpm was and will update the encoder steps as though a motor operated at that speed for the amount of loopinterval. 

It adds up to 10% error to mimic servo noise.  It doesn't do a great job of testing the PID settings (this would be tough) and doesn't verify that the encoder library is working.  But other than that I think this tests the rest of the code in the Firmware.

It closes #361
closed 2018-01-08 by BarbourSmith