MaslowCNC/Firmware #363: Create AVR Simulated Project for Virtualized Testing of Firmware

krkeegan, 2018-01-06:
SimAVR is an AVR simulator that can be used for testing of the firmware.  

https://github.com/buserror/simavr

I am sure other avr simulators are also available.

krkeegan, 2018-01-06:
I don't think so.  Processing looks like a robust means to communicate with an Arduino, but I don't think it would allow the simulation of the firmware on a PC.

idoc, 2018-01-18:
Simavr seems very promising, but has anyone gotten it to run any Arduino code successfully? I've spent about 15 hours so far trying to get anything to run for more than a few seconds, and it works, sort of, until I try to write to the serial port. Once I wrote more than a few hundred bytes to serial, the whole thing hangs. This happens even with trivial test programs. Maslow's firmware is chatty enough that it hangs before I even get a chance to open the serial monitor :)

I've opened an issue on the simavr repo. Hopefully it's just some boneheaded mistake  I'm making, and someone there can point it out.

idoc, 2018-02-15:
I have…

Something.

I (finally!) got most of the firmware running in simavr. It required much futzing with simavr code, but most of our code runs now, and I have  a simple recipe that others can copy-paste into a Mac or Linux terminal.

The issue is that it only runs if I disable the timer that runs runsOnATimer. Something about how we use interrupts in TimerOne.cpp just makes simavr hang after a few seconds.

I still think simavr has a decent chance of giving us a solution for trying out code without a spare Arduino, having unit tests that run automatically in a semi-realistic environment (a step towards a full CI environment?), and being able to attach gdb to the firmware and seeing what's going on inside.

I'm looking at two directions now:	

1. See if I can find an alternative timer implementation, one which works better in simavr. If I find one that works, I'll add an option in `Config.h` to use it when testing.

1. See if I can completely avoid using a timer in testing. Maybe call `runsOnATimer()` directly inside `loop()`? Maybe generate fake PID values whenever someone tries to read them?

Option 2 seems like it will require less reverse engineering of both simavr and Maslow, and will allow me to have something useful to share with everyone sooner.

To be honest, I only have a general idea of what `computePID()` is for. I'm guessing that in a simulator, it won't be a problem if it does run often enough, or if it returns inaccurate values, because there's no real sled or real motors to keep accurate track of.

Does this seem like a valid assumption that will give us useful simulation? Any advice would be much appreciated.

BarbourSmith, 2018-02-15:
@krkeegan what about the #define for putting the firmware in simulation mode, do you think that could help? I bet the issue has to do with trying to read from the pins which don't exist

Should I set up something like circle ci (https://circleci.com/pricing/#build-linux)? We could have it run every time something is pushed to master

krkeegan, 2018-02-16:
That is awesome.

Umm, I have no idea why the timer portion doesn't work.  Is there anyway to debug this?  I.e. can we test if the runsontimer works if it is empty?  It might be the writing to the pins that does it, that only happens here.

blurfl, 2018-02-16:
There's a newer version of the TimerOne library here:
[https://github.com/PaulStoffregen/TimerOne](https://github.com/PaulStoffregen/TimerOne)
I've tried it, it seems to work correctly with our current dev master. It might make a difference as its been thoroughly rewritten. (I'd like to entertain changing to this new version as it adds the Teensy 3.x to the systems it works with, but that's for a different discussion...)

idoc, 2018-02-16:
Thanks for the responses, guys!

@BarbourSmith You mean `FAKE_SERVO`? I'm using that in all my tests. It works great with my spare Arduino, which does not have a shield, without any other changes in the code. In simavr, though, it doesn't keep things from hanging.

@krkeegan Emptying out `runsOnATimer()` doesn't help. As far as I can tell, it never gets called anyway. I've only had partial success with connecting gdb to simavr (still working on perfecting that part...), so I'm relying on `Serial.print("I'm in runsontimer!")`, which I never see in the output.

Maybe you can save me some figuring out: Since we would always* use FAKE_SERVO in simavr, do I even need to worry about `runsOnATimer()`? Or does `FAKE_SERVO` make it redundant? I could just disable the timer when `FAKE_SERVO` is defined.

@blurfl That's encouraging! Let me try that and report back.

Oh and I think it would be great to set up a CI account. Circle seems nice, and I like https://travis-ci.com's open source plan, but I have no direct experience with either of them - we use non-free stuff at <dayjob>. Even if we can't quite iron out simavr's kinks, it would be nice to have PlatformIO build our binaries, run some basic tests, and give us official release binaries.

-----------
* Actually, simavr is flexible enough that we could program a simulated motor shield and get readings from that, but I don't think that will be worth our effort anytime soon.

BarbourSmith, 2018-02-16:
Ohhhh I like travis-ci's open source plan more than circle also! Plus they support python 👍 👍

idoc, 2018-02-18:
Behold!

Maslow in gdb.
[image]
I use plain terminal avr-gdb for most of my experimentation, but gdbgui makes for nicer screenshots :)

Ground Control still can't talk to simulated Maslow in any meaningful way, so there's much left to do, but now that breakpoints work properly, I feel like there's enough content to share and get some early feedback.

Next steps:
* PR to add a PlatformIO project. I find PlatformIO convenient in general, and for this project, having the binary always built in the same directory was a huge timesaver. I think this PR will also solve #365.
* PR with some code changes that were needed to run Maslow under simavr.
* PR with my simulow code.
* Add a wiki page with Mac and Linux instructions
* Figure out what's stopping GC from talking to simulow.

idoc, 2018-02-19:
@blurfl I tried the new TimerOne, with the same results. Still haven't been able to figure out why :(

I included a workaround for this in GH-396, which moves the call to runsOnATimer() into loop() when running in the simulator.

blurfl, 2018-02-19:
This is great work! 

> Next steps:
PR to add a PlatformIO project. I find PlatformIO convenient in general, and for this project, having the binary always built in the same directory was a huge timesaver. I think this PR will also solve #365.
PR with some code changes that were needed to run Maslow under simavr.
PR with my simulow code.
Add a wiki page with Mac and Linux instructions
Figure out what's stopping GC from talking to simulow.

Do you want a separate issue for these next steps? With an assignment?😁

idoc, 2018-02-25:
Aaaand done! At least, as requested here.
The simulator software is in place, and I've created preliminary [instructions](https://github.com/MaslowCNC/Firmware/wiki/Running-the-firmware-in-a-simulator).

I've marked the last PR as "Resolves #363", because I think there's a lot of debugging and testing that's already possible, but I'm not done working on this area until GC mistakes the simulated firmware for real thing.

blurfl, 2018-02-25:
Very well done!

commit 45d76224 (2018-02-27): Add simduino program for simulating and debugging Maslow firmware

Resolves #363
closed 2018-02-27 by BarbourSmith