MaslowCNC/Firmware #365: Feature Request: makefile build process of firmware (for non-x KivyPi setup)

ladams00, 2018-01-08:
I'm enamored how well GroundControl with KivyPi runs without Xorg on the Raspberry Pi.  Then I got to thinking, "I'm still going to need to use Xorg for compiling and uploading the Arduino's firmware with the IDE".  After reading the CLI section on arduino.cc, they present a simple makefile process to build and upload firmwares without the full blown IDE.  How cool would it be to include a makefile configuration that pulls the latest code from the git repo, build and uploads?  No need for Xorg :-)

As soon as the Maslow kit arrives in the mail, I'll get started on putting this makefile config together, unless someone beats me to it :-)

blurfl, 2018-01-08:
Welcome, it sounds like you've got the right spirit! 👍. FWIW, the Arduino in the kit is a clone of the Arduino Mega2560 from [Keyestudio.com](http://www.keyestudio.com/shop/board/mega.html). There's nothing special about it, and writing the firmware to any Arduino Mega2560 will work even without the kit. You can even run GroundControl and practice cutting files (if you click past the warnings that the chain settings are wrong). So, if you've got a Mega2560 laying around, you don;t have to wait to begin playing😁!

ladams00, 2018-01-08:
Hello @blurfl , Thank you for the advice and encouragement.  I don't have a MEGA2560, but just placed an order for one.  In the mean time, I'll work on getting the build running.  Thanks again!

blurfl, 2018-01-08:
👍  I looked at two Makefile approaches, both require the Arduino IDE to be installed. Would that violate the goal of the feature request?

ladams00, 2018-01-08:
Not at all.  Having the Arduino IDE installed is just fine.  It includes the CLI components (avr-gcc, avrdude), and the processing based IDE GUI front end.  Simply doing an "apt-get install arduino" takes care of installing it.  By going the makefile approach, not only does the firmware build and upload get a chance to be streamlined and automated, but we wont have to rely on Xorg for those tasks.  Seeing how KivyPi allows for GroundControl to run without Xorg reminded me how I love that omxplayer for Raspberry Pi takes a similar approach.

blurfl, 2018-01-10:
@ladams00, do you want this as an assignment, or would you want the 'help wanted' label on it?

ladams00, 2018-01-10:
Hey @blurfl , I suppose in the mean-time, I can take this on as an assignment. As a getting started task, I looked at the verbose build output of the standard GUI IDE and started to realize there is quite a bit that goes on during the build process.  Eventually, I think I may need some help.
```
/home/ladams/Downloads/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/ladams/Downloads/arduino-1.8.5/hardware -tools /home/ladams/Downloads/arduino-1.8.5/tools-builder -tools /home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -built-in-libraries /home/ladams/Downloads/arduino-1.8.5/libraries -libraries /home/ladams/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path /tmp/arduino_build_468928 -warnings=none -build-cache /tmp/arduino_cache_448500 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -verbose /home/ladams/git/Firmware/cnc_ctrl_v1/cnc_ctrl_v1.ino
/home/ladams/Downloads/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/ladams/Downloads/arduino-1.8.5/hardware -tools /home/ladams/Downloads/arduino-1.8.5/tools-builder -tools /home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -built-in-libraries /home/ladams/Downloads/arduino-1.8.5/libraries -libraries /home/ladams/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path /tmp/arduino_build_468928 -warnings=none -build-cache /tmp/arduino_cache_448500 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr -verbose /home/ladams/git/Firmware/cnc_ctrl_v1/cnc_ctrl_v1.ino
Using board 'mega' from platform in folder: /home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr
Detecting libraries used...
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_468928/sketch/cnc_ctrl_v1.ino.cpp" -o "/dev/null"
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/variants/mega" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/libraries/EEPROM/src" "/tmp/arduino_build_468928/sketch/cnc_ctrl_v1.ino.cpp" -o "/dev/null"
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/variants/mega" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/libraries/EEPROM/src" "-I/home/ladams/Downloads/arduino-1.8.5/libraries/Servo/src" "/tmp/arduino_build_468928/sketch/cnc_ctrl_v1.ino.cpp" -o "/dev/null"
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/variants/mega" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/libraries/EEPROM/src" "-I/home/ladams/Downloads/arduino-1.8.5/libraries/Servo/src" "/tmp/arduino_build_468928/sketch/Axis.cpp" -o "/dev/null"
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/variants/mega" "-I/home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/libraries/EEPROM/src" "-I/home/ladams/Downloads/arduino-1.8.5/libraries/Servo/src" "/tmp/arduino_build_468928/sketch/Encoder.cpp" -o "/dev/null"
[... cut ...]
"/home/ladams/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/tmp/arduino_build_468928/cnc_ctrl_v1.ino.elf" "/tmp/arduino_build_468928/cnc_ctrl_v1.ino.hex"
Using library EEPROM at version 2.0 in folder: /home/ladams/Downloads/arduino-1.8.5/hardware/arduino/avr/libraries/EEPROM 
Using library Servo at version 1.1.2 in folder: /home/ladams/Downloads/arduino-1.8.5/libraries/Servo 
Sketch uses 52564 bytes (20%) of program storage space. Maximum is 253952 bytes.
Global variables use 1850 bytes (22%) of dynamic memory, leaving 6342 bytes for local variables. Maximum is 8192 bytes.
```

ladams00, 2018-01-10:
One quick thing I wanted to add: Utilizing the packaged version of Arduino from Canonical's (Ubuntu) repository is horribly out-dated and the Maslow Firmware will not build correctly (the IDE complains about missing libraries...that aren't missing at all).  Using the tarball from arduino.cc is the way to go.

blurfl, 2018-01-10:
> Utilizing the packaged version of Arduino from Canonical's (Ubuntu) repository is horribly out-dated and the Maslow Firmware will not build correctly (the IDE complains about missing libraries...that aren't missing at all). Using the tarball from arduino.cc is the way to go.

That's a good insight. Why not bring it up in the forum and see what others have found? If others have struggled with this, it probably should go into the Linux instructions or the wiki.

idoc, 2018-02-19:
I just uploaded GH-395, which adds a PlatformIO project. PlatformIO (http://docs.platformio.org/en/latest/installation.html#installation-methods) lets you build the firmware and upload it to Maslow from the command line.

@ladams00 Does that cover everything you need for your project?

ladams00, 2018-02-19:
@idoc , this works beautifully!  Here's what I've done step by step on the kivypie with Ground Control:

- Checkout your fork and switch branches
`git clone https://github.com/idoc/Firmware && git checkout remotes/origin/platformio-project`
- Install PlatformIO
`sudo pip install -U platformio`
- Change dir and build Firmware with PlatformIO
`cd Firmware && pio run`
- Flash Arduino with hex file
`avrdude -C /usr/share/arduino/hardware/tools/avrdude.conf -v -v -p atmega2560 -c wiring -P /dev/ttyACM0 -D -U  #flash:w:.pioenvs/megaatmega2560/firmware.hex:i`

Using this procedure, there is zero need for the full blown Arduino IDE (and Xorg) and allows a Kivypie setup to operate 100% on it's own.  This flashing procedure could be placed into a shell script to include checking out the latest releases via git as well.

Nicely done!  This issue can be closed.

idoc, 2018-02-22:
@ladams00  Glad it's working for you!

Note that if your Maslow is the only serial device connected to your RasPi, you can slightly simplify your upload command by having PlatformIO do it for you:
`platformio run -t upload`

ladams00, 2018-02-22:
@idoc , I just ran 'pio run -t upload' and noticed for each build target (both megaatmega2560 and simavr), get uploaded to the Arduino after each linking operation (and simavr being second).  Based on other conversation threads relating to PlatformIO and simavr, I'm betting the builds are a bit different regarding on TimerOne, etc?  Here's the output below:
```
sysop@kivypie:~/git/Firmware$ pio run -t upload
[Thu Feb 22 17:34:03 2018] Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------
PackageManager: Installing tool-avrdude @ ~1.60300.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
Converting cnc_ctrl_v1.ino
Collected 24 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <EEPROM> v2.0
|-- <Servo> v1.1.2
Compiling .pioenvs/megaatmega2560/src/cnc_ctrl_v1.ino.o
In file included from cnc_ctrl_v1/Maslow.h:26:0,
from /home/sysop/git/Firmware/cnc_ctrl_v1/cnc_ctrl_v1.ino:15:
/home/sysop/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/EEPROM/src/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
static EEPROMClass EEPROM;
^
Linking .pioenvs/megaatmega2560/firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
54688	    154	   1697	  56539	   dcdb	.pioenvs/megaatmega2560/firmware.elf
Building .pioenvs/megaatmega2560/firmware.hex
Looking for upload port...

Warning! Please install `99-platformio-udev.rules` and check that your board's PID and VID are listed in the rules.
https://raw.githubusercontent.com/platformio/platformio/develop/scripts/99-platformio-udev.rules
Auto-detected: /dev/ttyACM0
Uploading .pioenvs/megaatmega2560/firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file ".pioenvs/megaatmega2560/firmware.hex"
avrdude: writing flash (54842 bytes):

Writing | ################################################## | 100% 9.69s

avrdude: 54842 bytes of flash written
avrdude: verifying flash memory against .pioenvs/megaatmega2560/firmware.hex:
avrdude: load data flash data from input file .pioenvs/megaatmega2560/firmware.hex:
avrdude: input file .pioenvs/megaatmega2560/firmware.hex contains 54842 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.05s

avrdude: verifying ...
avrdude: 54842 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

========================= [SUCCESS] Took 35.85 seconds =========================

[Thu Feb 22 17:34:39 2018] Processing simavr (extra_scripts: simavr_env_for_platformio.py; build_flags: -g -O0; build_unflags: -Os; platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting cnc_ctrl_v1.ino
Collected 24 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Servo> v1.1.2
|-- <EEPROM> v2.0
Compiling .pioenvs/simavr/src/cnc_ctrl_v1.ino.o
[... 31 lines cut ...]
Writing | ################################################## | 100% 19.06s

avrdude: 108150 bytes of flash written
avrdude: verifying flash memory against .pioenvs/simavr/firmware.hex:
avrdude: load data flash data from input file .pioenvs/simavr/firmware.hex:
avrdude: input file .pioenvs/simavr/firmware.hex contains 108150 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 13.87s

avrdude: verifying ...
avrdude: 108150 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

========================= [SUCCESS] Took 46.17 seconds =========================

================================== [SUMMARY] ==================================
Environment megaatmega2560	[SUCCESS]
Environment simavr        	[SUCCESS]
========================= [SUCCESS] Took 82.03 seconds =========================
sysop@kivypie:~/git/Firmware$ 
```

idoc, 2018-02-22:
@ladams00 Oh right, I forgot GH-399 has not been merged yet. Once that's in, the simavr environment will not try to upload. For now, you'll need to run `pio run -t upload -e megaatmega2560`.

idoc, 2018-02-22:
@ladams00 It's merged. Hope it works for you now!

ladams00, 2018-02-22:
@idoc , Thanks!  I'll try it later this evening :+1:

ladams00, 2018-02-23:
@idoc, Works great :100: Thank you again!

PR #395 (2018-02-19): Add a PlatformIO project
This project builds for two environments:
megaatmega2560: This is the standard Arduino Mega 2560 setup for
PlatformIO. It uses the default flags for PlatformIO, and builds an
optimized, stripped binary. Good for building firmware for production
use.
simavr: Disable all optimizations, leave debugging symbols, and #define
SIMAVR. SIMAVR doesn't do anything yet, but it will enable some
simavr-specific behavior in a separate commit.

I created this project as prep for #363, but it can also help simplify
setup for users of the PlatformIO IDE, or someone setting up a CI
envi [... cut]
commit 1454ee9f (2018-02-19): Add a PlatformIO project

This project builds for two environments:
megaatmega2560: This is the standard Arduino Mega 2560 setup for
PlatformIO. It uses the default flags for PlatformIO, and builds an
optimized, stripped binary. Good for building firmware for production
use.
simavr: Disable all optimizations, leave debugging symbols, and #define
SIMAVR. SIMAVR doesn't do anything yet, but it will enable some
simavr-specific behavior in a separate commit.

I created this project as prep for #365, but it can also help simplify
setup for users of the PlatformIO IDE, or someone setting up a CI
env [... cut]
closed 2018-02-23 by ladams00