MaslowCNC/Firmware #366: Buffer overflow when using manual z-axis

BarbourSmith, 2018-01-08:
As described in this forum thread:  https://forums.maslowcnc.com/t/caught-buffer-overflow-in-the-act/1798/1

The internal buffer overflows when using the manual z-axis

BarbourSmith, 2018-01-08:
After some poking around it looks like the number of lines stored in the buffer grows by one every time the machine pauses, so one extra line is being sent...I bet this has to do with the fact that more information than normal is being returned because a request to have the user set the depth is also being sent.

blurfl, 2018-01-08:
I saw some extra linefeeds - are we sending two with the request, or failing to purge the li

BarbourSmith, 2018-01-08:
Good point! I will check into that

BarbourSmith, 2018-01-08:
Here the log for the section where it happening with some debug code:

```
G1 Z-0.5 F30

Lines in buffer: 2

BEGIN PAUSE

Message: Please adjust Z-Axis to a depth of -0.50 in

Middle of pause

Buffer Used: 49

Buffer Number of Lines: 2

Buffer Begin: 23

Buffer End: 72

Buffer Contents: G1 X14.2454 Y-0.3842 F60  \nG1 X12.492 Y-0.5376  \n(End of Buffer)

Maslow Paused

Sent: ~ 
ok

Send another line

End of pause

ok

Sent: G1 X12.3548 Y-0.4045  
Send another line

G1 X14.2454 Y-0.3842 F60

Lines in buffer: 3
```

We can see that with each cycle of the z-axis we get one more line in the buffer

BarbourSmith, 2018-01-08:
I think that the issue is here where at the beginning and end of the pause we are sending an 'ok'

```
Sent: ~ 
ok

Send another line

End of pause

ok
```

blurfl, 2018-01-09:
So we've sent two 'ok's but only handled one line in the buffer, yes? Isn't 'ok' permission to send another line?

krkeegan, 2018-01-09:
The firmware will respond with an 'ok' to the ~ command.  Does GC not expect one there?  I

davidelang, 2018-01-09:
On Mon, 8 Jan 2018, Scott Smith wrote:

> So we've sent two 'ok's but only handled one line in the buffer, yes? Isn't 'ok' permission to send another line?

Yes it is (we never did get the bytecounting buffering working right)

blurfl, 2018-01-09:
The log line with the buffer contents ar the overflow shows  '\r' characters (I thought we stripped those?) and one place with '\r\n\n':

> Buffer Contents: G3 X8.7285 Y3.6627 I-0.125 J0\r \n \nG1 X2.242 Y8.5469\r \nG3 X2.1668 Y8.572 I-0.0752 J-0.0999\r \nG1 X0.67 Y8.572\r \nG3 X0.545 Y8.(End of Buffer)

Is that the extra line we're talking about?

BarbourSmith, 2018-01-09:
> The firmware will respond with an 'ok' to the ~ command. Does GC not expect one there? 

I guess not, it seems like that was one extra 'ok'. I know you moved things around recently @krkeegan but the issue was totally my issue that just got moved to a new place...speaking of which I'm loving the new organization system 👍 👍

BarbourSmith, 2018-01-09:
I think we need the '\r\n' characters because they are used to mark where one line ends and the next begins

krkeegan, 2018-01-10:
Nice, I am glad we caught all this.  I was never very happy with all the extra ok commands flying around.

BarbourSmith, 2018-01-10:
Me too...This has been around forever

BarbourSmith, 2018-01-16:
Unfortunately, this still seems to be an issue. I can replicate it on master using this file: 
[sledC-30.txt](https://github.com/MaslowCNC/Firmware/files/1636989/sledC-30.txt)

BarbourSmith, 2018-01-16:
Original fix was in #368  which looks like it should work when I go back and look at it 😐

BarbourSmith, 2018-01-16:
It looks like it works because it does work! 👍 👍  I was testing on a branch which wasn't up to date 😒 

The problem is still solved, sorry for being foolish and not noticing at first 🙄

closed 2018-01-16 by BarbourSmith

krkeegan, 2018-01-16:
That's good, my brain hurt the last time we had to solve this.

commit 5da45297 (2018-01-09): Catch extra 'ok' message sent on  manual z-axis move

The firmware was sending an extra 'ok' status when paused prompting the machine to gain one extra line in the buffer with each z-axis adjustment eventually overflowing the buffer.

The solution is to remove the extra command which does not seem to be needed for operation with our without the z-axis.

The pause command does not seem to be effected.

Fixes #366
PR #368 (2018-01-10): Catch Line Ending on Quick Commands; Remove Watchdog
Need to discard any line endings sent after a quick command
otherwise they get interpreted as a blank line and an OK is
sent.  I am not super thrilled with how I wrote this, but I
think it is reasonably compact and logical.

REMOVE WATCHDOG.  This was also causing an issue when the
machine paused itself waiting for a manual Z change.  If
the buffer was empty, it would send an OK requesting a new
line.  We just need to be careful now to make sure that all
OKs are properly sent.

This along with changes to GC closes #366