MaslowCNC/Firmware #370: Forward Kinematics Improvements

krkeegan, 2018-01-16:
I have a few kinematics questions.  I don't claim to understand all of this math very well, so any guidance would be appreciated.

First, I noticed an oddity that I don't know if I caused or if this has always been there.  If when set to quadrilateral kinematics I 

1. set the left chain to 0 and the right chain to 1650, and $K (recompute kinematics) I get an error (unable to find sled position).
2. set the left chain to 1650 and rerun $K. I still get an error.
3. restart GC and the sled position is properly calculated.

I chased this down and the issue is that Phi eventually becomes nan and this is only reset on a complete restart.  Adding the initial declared value of Phi = -0.2 to recomputeKinematics fixes the issue.  Has this been an oversight?  What does Phi represent?  Will resetting it cause an issue?

Second, I thought it would be nice to not have to call $K anymore and to update the kinematics whenever a relevant setting is changed.  The issue is that forward kinematics can take up to 6 seconds to fail out and with ~6 settings that affect kinematics this sucks a bit.

I added in an additional threshold in forward, if the guessed chain length exceeds the 11 foot chain length then the forward calculation ends early.  This is about a 10x savings in time, and means that forward fails in about 700 milliseconds (much more reasonable).  The calculations seems to work fine in all other respects.

I would make the chain length a user setting so that it isn't hardcoded at 11 feet.  Does anyone see any issues with this approach?  In watching the forward calculations I didn't see any instances where the guessed chain length gets larger than the allowed length only to turn around and become smaller again.

blurfl, 2018-01-16:
>  chain length exceeds the 11 foot chain length 
I've added a couple feet to my chains, my motor spacing is 11' (3364mm) and the motor y height is 1.5' (464mm, pretty standard).
 Is there something I could test for this?

davidelang, 2018-01-16:
the problem is the chain length of 0, that's not valid.

the two chain lengths are not going to work if they add up to less than the 
distance between the motors (well, technically it could work if chain lengths + 
distance between chain mounts > distance between the motors, but the motors 
cannot pull things that high)

I'm nervous about adding max chain length calculations in, I expect that it 
will cause more problems than it solves.

If you are trying to figure out where you are based on the current chain 
lengths, you need to assume that those chain lengths are valid, even if one is > 
11 ft.

David Lang

On Tue, 16 Jan 2018, Kevin Robert Keegan wrote:

> Date: Tue, 16 Jan 2018 21:10:13 +0000 (UTC)
> From: Kevin Robert Keegan <notifications@github.com>
> Reply-To: MaslowCNC/Firmware
>     <reply+0021b24f990ae28622c1994e1de86b7ef039f0d5ccd27cfb92cf0000000116762c3
>     492a169ce113aa94a@reply.github.com>
> To: MaslowCNC/Firmware <Firmware@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [MaslowCNC/Firmware] Forward Kinematics Improvements (#370)
> 
> I have a few kinematics questions.  I don't claim to understand all of this math very well, so any guidance would be appreciated.
>
> First, I noticed an oddity that I don't know if I caused or if this has always been there.  If when set to quadrilateral kinematics I
>
> 1. set the left chain to 0 and the right chain to 1650, and $K (recompute kinematics) I get an error (unable to find sled position).
> 2. set the left chain to 1650 and rerun $K. I still get an error.
> 3. restart GC and the sled position is properly calculated.
>
> I chased this down and the issue is that Phi eventually becomes nan and this is only reset on a complete restart.  Adding the initial declared value of Phi = -0.2 to recomputeKinematics fixes the issue.  Has this been an oversight?  What does Phi represent?  Will resetting it cause an issue?
>
> Second, I thought it would be nice to not have to call $K anymore and to update the kinematics whenever a relevant setting is changed.  The issue is that forward kinematics can take up to 6 seconds to fail out and with ~6 settings that affect kinematics this sucks a bit.
>
> I added in an additional threshold in forward, if the guessed chain length exceeds the 11 foot chain length then the forward calculation ends early.  This is about a 10x savings in time, and means that forward fails in about 700 milliseconds (much more reasonable).  The calculations seems to work fine in all other respects.
>
> I would make the chain length a user setting so that it isn't hardcoded at 11 feet.  Does anyone see any issues with this approach?  In watching the forward calculations I didn't see any instances where the guessed chain length gets larger than the allowed length only to turn around and become smaller again.
>
>
>
>

krkeegan, 2018-02-23:
I am going to go ahead and close this.  We have been operating with both the change to Phi and the chain length fix for two weeks without issue.  Apologies I should have mentioned this issue in the commit so that it would show up here.  

It was #384.

closed 2018-02-23 by krkeegan