MaslowCNC/Firmware #373: If you move the Z axis without a z axis motor plugged in, L moves instead

thormj, 2018-01-29:
No error message on groundcontrol, so I'm not sure if it knows what's going on...  Now that I have it plugged, I hear the L motor being driven (but the Z motor is doing what it's supposed to).

BarbourSmith, 2018-01-29:
I think that both motors will power up when the z-axis is command to move (well actually all three motors power up any time one moves). The L and R should just be holding position instead of moving. Do you see actual movement or is it only the noise?

blurfl, 2018-01-30:
On my machine although the L motor makes noise, the X and Y coordinate values do not change, indicating that the L and R motors have not moved.

thormj, 2018-01-30:
Agreed... sorry about the misfire.  I was trying to see what was wrong with my z-axis and started troubleshooting by ear (no motion, but power gets applied to L so I can hear it)... 
It took me a while to figure out that I forgot to plug it in.

blurfl, 2018-01-30:
I’ve been there 😁👍. You can close it if you’re satisfied.

closed 2018-01-30 by thormj

BarbourSmith, 2018-01-30:
I think @blurfl has a genius fix for this in the works 😉 so hopefully they will sound the same soon