MaslowCNC/Firmware #379: Machine won't start again after pressing stop

BarbourSmith, 2018-02-01:
Moved from GC issue [#603](https://github.com/MaslowCNC/GroundControl/issues/603)

For some reason my machine has stopped starting again after the stop button is pressed. It will stop immediately, but not move again until GC is restarted. 

This is a new behavior.

BarbourSmith, 2018-02-01:
Or maybe it's that we're getting one 'OK' instead of two.

From the log:

`Sent: !
ok
`

But when we print the amount of space GC thinks is available in the buffer it looks like this:

[image]

So there's `236 `spaces in the buffer because the line of gcode hasn't been processed yet, then there are `234` spaces because we've sent the "`!\n`" command, then we go back to `236` when it is marked that the stop command has been processed...so do we need to send an extra "OK" after stops are processed to indicate that the extra line has been cleared from the buffer? It has been cleared.

BarbourSmith, 2018-02-01:
I think this is related to #366

BarbourSmith, 2018-02-01:
Here's what I'm seeing is happening. 

The '!\n' is sent, and we immediately get an 'OK' status to indicate that it has been received from GCode.cpp line 51. Within the `coordinatedMove` function in Motion.cpp line 163 `if (sys.stop){return 1;}` is triggered and the machine stops right away (yay 🎉 ). 

The issue comes up because that return kicks us back out GCode.cpp line 663 where coordinated move was called, and nothing happens there to indicate that the line of gcode is complete `coordinatedMove(xgoto, ygoto, zgoto, 1000);.`

We then drop back up another to line 339 `G1(gcodeLine, gNumber);` and eventually kick out to line #594 which is `if (!sys.stop){reportStatusMessage(status);}` so we don't send the OK message there.

As far as I can tell !sys.stop is never cleared, right?

Is it safe to report the 'OK' after the stop command has been processed? I think it is because we've stopped sending gcode on the Ground Control side at that point.

I'm a little out of my element here @krkeegan do you have thoughts on what the best solution is?

krkeegan, 2018-02-01:
if sys.stop is true it causes the while loop in the main loop in cnc_ctrl_v1.ino at line 86 to bail out.  At which point we reset everything and clear the sys.stop flag.

@bar is this all on the current master versions of the firmware and GC?  And this is happening 100% of the time or is there some sequence I should try to reproduce?

blurfl, 2018-02-01:
I see this issue with versions 1.05, every time I use the Stop button. Stopping while using the move arrows will show it.

krkeegan, 2018-02-01:
ugh, I may have broken my cheap testing arduino or something.  It doesn't seem to be responding at the moment.

BarbourSmith, 2018-02-01:
I've had good luck bringing arduinos back to life by pressing the black reset button on them right before the firmware starts to upload. If I remember right pressing that button gives you a 3 second window before the code starts to run again

krkeegan, 2018-02-01:
My arduino isn't even recognized as a thing in windows.  I will check it out on a proper linux machine tonight.  But in the interim, I can't even simulate what you guys are seeing.

BarbourSmith, 2018-02-01:
Darn! That sounds pretty gone 😞 

You want me to send you a new one or two?

bar, 2018-02-01:
@krkeegan I think you might be confused

BarbourSmith, 2018-02-01:
Sorry @bar 

BTW is your name actually Bar? I've never met another one

BarbourSmith, 2018-02-01:
Moved back to the GC issue

closed 2018-02-01 by BarbourSmith