MaslowCNC/Firmware #381: Allow chains to connect to sled from top or bottom of motor sprockets

reecej, 2018-02-02:
I'm working on a PR to allow the user to specify whether the chains attach to the sled from the top or bottom of the motor sprockets, essentially allowing the user to choose which way the chains wrap around the sprockets. But I have some questions on how best to do this.

The primary issue is that the direction each motor needs to spin to feed out a positive length of chain is different for each configuration. As an example, for the left motor to feed out chain with the chain exiting on the top of the sprocket, it would need to spin clockwise. If the chain direction were reversed such that the chain exited from the bottom of the sprocket to the sled, then the motor would need to spin counterclockwise to feed out positive chain.

I think the cleanest solution to this is to update the axis configuration for the right and left motors, based on which direction the user specifies the chain is wrapped around the sprockets to the sled. Would the best method to accomplish this be to swap the IN and ENCODER variables in the left/right axis setup parameters in System.cpp (lines 154-155)?

krkeegan, 2018-02-02:
I think what you are describing should work, and is probably the best way to do it.  At least from the motor/encoder perspective.

You should also look at the kinematics calculations as well.

blurfl, 2018-02-02:
That certainly works, jsut ran it. Nicely and cleanly done! 👍

reecej, 2018-02-02:
I agree, the kinematics will need to be updated as well.

I just wanted to ensure that I did this in the most effective way possible. At first I thought about trying to use negative chain lengths for the chain on bottom case, but it just gets messy fast.

Thanks @blurfl! As a follow-up question, do I need to swap the IN and ENCODER variables for the left and right axes, or just the IN values?

krkeegan, 2018-02-02:
Both, otherwise the encoder values will be negative.

reecej, 2018-02-02:
Perfect! Thank you guys for your help!

reecej, 2018-02-02:
One additional question I have. I have updated the triangular kinematics to account for the chain wrapping either direction around the sprocket, and am now trying to do the same for the quadrilateral kinematics. In looking at the calculations for the chain lengths in Kinematics.cpp I see R * Gamma and R * Lambda, which look to be the parts that account for the chain wrapping around the sprocket. My question is, do I need to modify the R * TanGamma and R * TanLambda values? I'm not sure what they do and if they are related to the chain wrapping around the sprocket.

davidelang, 2018-02-02:
I would not bother with quad kinematics at all. If someone is modifying their 
machine to have a top beam to support the reverse chain rotation, they are 
almost always going to have switched to some form of a triangular sled mount

and if they haven't, they really should :-)

reecej, 2018-02-02:
That's what I was thinking also, but wanted to check with others to make sure. I see two situations:

1) A new user using the new frame design with a top beam
2) An existing user adding a top beam to increase accuracy

Case 1 will use triangular kinematics out of the box, so that's not a problem. For case 2, if they are that focused on accuracy I would think they have already installed some sort of linkage kit too.

The reason I ask is that without updates to the quadrilateral kinematics, using the chains in the reverse direction will slightly decrease accuracy, but if it's a rare use case then it likely isn't a problem.

blurfl, 2018-02-02:
Updating the quad kinematics could be opened as a separate issue, to keep it on the books. If there is enough interest then the issue could be addressed.

BarbourSmith, 2018-02-02:
Thanks for tackling this! 

I ran a test cut with the chains wrapping the other way and it seems to work really well. Flipping both the IN and Encoder lines worked properly, the only issue is the part came out distorted so I am excited for the new kinematics.

[image]

blurfl, 2018-02-14:
I think this was accomplished in PR#607, wasn't it? OK to close this?

reecej, 2018-02-14:
Yes, please! I'll open a separate issue clarifying that the chain on bottom does not account for chain on the sprocket properly using quadrilateral kinematics, but otherwise this is fine to close. Thanks for catching that!

closed 2018-02-14 by blurfl