MaslowCNC/Firmware #388: Max Feed is cut in Half in Some Work Area Locations

krkeegan, 2018-02-08:
I am a bit stumped.  There are some areas on the work surface where the feed rate of the sled is cut in half.

To see this try (I am working with a quadrilateral setup that is rather standard):
1. Move the sled to the absolute position X: -300mm, Y:-430mm
2. Move right 100mm using the jog buttons, you will see and hear the sled moving slowly. 500mm/min
3. Move back left 100mm
4. Move left again 100mm. Somwhere around -320 you will hear and see a speed jump to 1000 mm/min

I have put in some debugging locally in the coordinatedMove function, and this isn't a feed rate or step distance issue.  These things remains correct.  It looks like something is happening inside the while loop?  Is something like the inverse calculation all of a sudden taking longer?

krkeegan, 2018-02-08:
OK, I can confirm that the inverse calculation is taking twice as long as normal in this location.  I also believe I see it in a few other locations on the work area as well.

So I think something we added to the kinematics recently that seems to take very long to calculate in certain positions.

krkeegan, 2018-02-08:
The issue seems to be that the while loop within the quadrilateralInverse normally takes 1 try to get within the acceptable range, but in this area it is taking 2 tries.

This looks like something that may go farther back than I realize and is probably something I did, any idea?

krkeegan, 2018-02-09:
This exists all the way back to v0.98, so this isn't a new creation.

I don't know is this something we are just stuck with?

Anyone have any ideas about what in the calculation causes inverse to run twice or anyway to store or save some value that might help us perform the calculation in a single pass?

BarbourSmith, 2018-02-12:
I thought we were using a timed delay function to make it so that the speed was based on the elapsed time and not on the time to converge? Is that not the case?

krkeegan, 2018-02-12:
So, the PID loop runs at specific times 10 milliseconds I think.  The inverse function loops until it reaches what looks like some maximum error.  The problem is that the inverse function takes about 6 milliseconds to run once, so it if has to loop again it takes longer than the pid loop.  The result is that we only send movement commands every other time.  Luckily, it looks like on most of the work surface the inverse calculation only needs to run once.

This only happens with the quadrilateral inverse.  The triangular is much faster.

reecej, 2018-02-12:
I've been looking through this, trying to better understand the quadrilateral kinematics to help troubleshoot this issue. If it happens around (-300,-430), then I presume it also happens around (300,-430)? Does it happen the entire way between x=-300 to x=300?

One thought I had was that since the quadrilateral kinematics are temporarily assuming x=0 is shifted to the centerline of the left motor axis, combined with the mirroring of areas on the right side of the bed, the area between x=-300 and x=300 represent the "furthest" away points, from the perspective of the algorithm. I wonder if it's just running out of cycles before it gets there.

davidelang, 2018-02-12:
near the middle you can run into problem cases, especially if some machine 
dimensions are slightly wrong.

the fix is to get a triangulation kit.
k

reecej, 2018-02-12:
I agree that I think the project is generally heading in the direction deprecating the quadrilateral kinematics, but I wouldn't say we are there yet. Before that happens I would expect at minimum something like a validated linkage kit Gcode which could be cut out using the quadrilateral kinematics, along with further quantitative analysis of the performance of the kit.

Until then, I think this needs to be resolved.

BarbourSmith, 2018-02-20:
I believe that this issue is related to the conversation going on in the forums here...and it seems that the problem is not limited to quadrilateral kinematics, it seems to effect the triangular kinematics as well.

https://forums.maslowcnc.com/t/velocity-slow-down-and-arced-cut/2413/2

reecej, 2018-02-20:
I thought the same thing at first, but the more I looked at it the more I'm wondering if it is the same. So far it seems that the users affected are operating in the high-tension portions of the bed area with the motors operating at the closest points to the stall torque, and the rest of the bed seems to be working fine. It very well may be related, but I am not yet convinced.

BarbourSmith, 2018-02-20:
Are we sure the slow down is limited to the quadrilateral kinematics? I think I remember seeing it on triangular too, but I'm not sure. I've got all my stuff taken apart right now to re-write the assembly instructions so I'm a little bit useless as far as testing things goes. As soon as I get the calibration process updated for the new frame design I'll be ready to jump back in.

reecej, 2018-02-20:
That's a good question. @krkeegan, I'm assuming that you are using quadrilateral kinematics and have observed the slowness issue. Do you have the ability to test with triangular kinematics?

If not, maybe we could see if @blurfl has input. It sounds like he has a lot of milage one his setup with triangular kinematics. It would be interesting to see if he could repeat the sequence that caused the slowdown, per this bug:

1. Move the sled to the absolute position X: -300mm, Y:-430mm
2. Move right 100mm using the jog buttons, you will see and hear the sled moving slowly. 500mm/min
3. Move back left 100mm
4. Move left again 100mm. Somwhere around -320 you will hear and see a speed jump to 1000 mm/min

reecej, 2018-02-20:
Or @BarbourSmith, if you have one of the February kits ready to ship I could give it a try too! ;-)

BarbourSmith, 2018-02-20:
I wish! They are all in the middle of the ocean 🙄 

I am going to give it a go on my setup and see what I can find out. It's not calibrated but it should still show us something

BarbourSmith, 2018-02-20:
In quadrilateral mode I'm seeing the velocity drop to 500mm/min from 1000mm/min at around x = -820mm, y= 0 

In triangular mode I am seeing the velocity stay constant all the way out to the edge, so I think my memory is wrong and it is only an issue when using quadrilateral kinematics

reecej, 2018-02-20:
That matches my understanding of this bug. It seemed like it was related to the inverse calculation specifically for quadrilateral kinematics, based on the iterative algorithm not being able to complete initially for certain areas of the bed (specifically the center it seems).

krkeegan, 2018-02-22:
I am pretty sure that this error goes way way back to pre v0.90 at least.  I only noticed it because of the new addition of the velocity read-out in GC.  So people have been living with this for a long time without issue.

While unfortunate, I don't think this bug has any major side effects other than taking longer to do something.  The way the bug manifests itself, the cuts are still performed correctly, just taking twice as long.  

There are a couple solutions I can think of:
1. Bump up the allowable error, I don't know how high this would have to be to avoid this on a stock setup.
2. Increase the loop interval, increasing it 40% would probably solve the issue, but would make for rougher cuts overall.
3. Figure out someway to increase the efficiency of the inverse calculation, I have no ideas for this.

Personally, I am inclined to mark this as a known issue that isn't planned to be fixed.  The Triangular setup seems like it is our eventual future, and other than irritation, this bug doesn't affect the "performance" of the machine.

BarbourSmith, 2018-02-23:
Well summarized. I agree wholeheartedly.

krkeegan, 2018-02-23:
Alright, closing for now, this can be reopened if this issue becomes larger or if someone has a hankering for solving it.

closed 2018-02-23 by krkeegan