MaslowCNC/Firmware #391: rename 'zAxisAuto' to 'spindleAutomate' to reflect its true use

blurfl, 2018-02-16:
PR #328 added a setting to enable/disable spindle control. The setting was named 'zAxisAuto' which seems confusing as it is being used as an on/off control for spindle power. Should we change the name and accompanying text in Report.cpp to 'spindleAutomation' to better reflect its use? Note that the change in GC that accomanied #328 was PR #458.

blurfl, 2018-02-16:
On further thought, this setting chooses between 'on/off' and 'servo-control' modes. Is there a better name than 'spindleAutomate' to convey this?

BarbourSmith, 2018-02-16:
Great point...I wouldn't have known what that setting did from reading the name. It's a tough thing to come up with a name for, I keep writing something and then deleting it. 

What about "spindle servo mode" or something and then making it clear in the text what the two options are. It's a pretty technical thing to understand.

blurfl, 2018-02-19:
Addressed in PR #392

closed 2018-02-19 by blurfl