MaslowCNC/Firmware #398: There is a calibration issue which is new since 1.06

BarbourSmith, 2018-02-21:
Running the attached file results in a curved line instead of a straight one. The issue is solved by reverting to firmware 1.03 so we know the issue is new.

[ttrapp2.zip](https://github.com/MaslowCNC/Firmware/files/1742502/ttrapp2.zip)

The original issue and information can be found in this forum thread:

https://forums.maslowcnc.com/t/velocity-slow-down-and-arced-cut/2413/48

I'm a little stumped on this one so it might be a good candidate for git-bisect to start narrowing down what is going on

krkeegan, 2018-02-22:
I read through the forum, has anyone else other than clintloggins been able to reproduce this error?  @BarbourSmith have you been able to reproduce this error?

BarbourSmith, 2018-02-23:
I believe that @blurfl was able to recreate it. I've been bogged down in updating the calibration process which has meant that my machine has not been a good test platform since it's calibration has been off. I'm hoping to wrap that up tomorrow and be able to test then.

GeroBH, 2018-03-03:
I cut it The outline only) today with 1.07. No curve visable
https://discourse-cdn-sjc2.com/standard11/uploads/maslowcnc/optimized/2X/8/84a8b20a28af176793f6704b7e08f9d925298905_1_282x500.jpg

BarbourSmith, 2018-03-03:
That is fantastic! 

It makes me nervous that we don't know what was causing the issue or what made it go away

reecej, 2018-03-03:
I agree. This has been a strange one, but I'm glad it is working better. I don't see anything in the changes for v1.07 that would jump out at me as resolving this.