MaslowCNC/Firmware #407: What to Replace B Commands With?

krkeegan, 2018-02-27:
So, the B commands like B05 (which requests the firmware version) have to go if we want to be compliant with GCode standards and things like GRBL.  I think the right thing is to move to something starting with $, because this would make it easy to allow the maslow specific commands to still be executed if the machine was in an alarm state.

My question, is what to replace them with.  I started coding up something in the form of:

`$MSLW4=Xxx
$MSLW5=Xxx`

Wherein the digit after MSLW denotes the command. We could do something like $M4 to save some typing of letters as well if people prefer that, GRBL doesn't currently use $M commands.

In this design, all commands would only accept one parameter after the equals sign.  This works for nearly all the commands, some of the current commands would be split in to (left and right) to allow for this.  The ones that don't are my own PID testing commands, which can probably be split into a number of commands and simplified.

Any thoughts or better ideas?

blurfl, 2018-02-27:
I lean toward the single letter option, less serial overhead.

BarbourSmith, 2018-02-27:
I think that sounds well thought out and sound. I miss being able to specify both a L and R value, but it's no great loss

krkeegan, 2018-02-27:
Agree I was kinda bummed by the L&R thing as well.  I will keep them in order so for example L would be even and R odd.

davidelang, 2018-02-27:
I think that where possible we should use exactly what grbl does, with the 
intent of replacing our parsing and planning logic with a fork of grbl.